PhotonVision/photonvision #608: Utilize NT4 Timestamps

gerth2, 2022-11-21:
NT4 adds the capiability for a coprocessor to sync up with the Rio's `Timer.getFPGATimestamp()` via https://www.geeksforgeeks.org/cristians-algorithm/

In NT3, this behavior is approximated by reporting the processing latency in NT, and assuming the network transit time is negligible. 

With NT4, instead, we can simply record the timestamp when the image is read from the camera, and carry it through to timestamp the results as they are submitted.

In photonlib, expose the timestamp of observation in the domain of `Timer.getFPGATimestamp()`.

Update examples to show using this timestamp for latency comp in the WPILib Kalman Fitlers, rather than latency.

Retaining latency is probably a good idea for debug purposes. Users will still want to tune pipelines to minimize it.

mcm001, 2022-12-17:
So just calling Timer.getFPGATimestamp() in photon is sufficient? Or how do we access the time in robot time base?

mcm001, 2025-03-20:
OBE, implemented

closed 2025-03-20 by mcm001