PhotonVision/photonvision #610: mDNS issues after beta-3

bznerd, 2022-11-24:
With a RPI 3B+ with a SnakeEyes Hat, after a few minutes running beta 6 or 5, the pi becomes unreachable via ssh or dashboard at photonvision.local

After plugging in a monitor to watch syslog it looks like the pi is withdrawing from mDNS.
After 5-10 minutes or so it will reconnect but then disconnect again after a few minutes.

It appears to cycle through joining the mDNS group and leaving it.

Here are the relevant lines of /var/log/syslog
```
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2209] policy: auto-activating connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2266] device (eth0): Activation: starting connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2289] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2369] manager: NetworkManager state is now CONNECTING
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2415] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2527] device (eth0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:26:17 photonvision NetworkManager[330]: <info>  [1666139177.2571] dhcp4 (eth0): activation: beginning transaction (timeout in 45 seconds)
Oct 18 20:26:17 photonvision avahi-daemon[5632]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::ac7e:36ab:5043:10a1.
Oct 18 20:26:17 photonvision avahi-daemon[5632]: New relevant interface eth0.IPv6 for mDNS.
Oct 18 20:26:17 photonvision avahi-daemon[5632]: Registering new address record for fe80::ac7e:36ab:5043:10a1 on eth0.*.
Oct 18 20:27:02 photonvision NetworkManager[330]: <warn>  [1666139222.2184] dhcp4 (eth0): request timed out
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2186] dhcp4 (eth0): state changed unknown -> timeout
Oct 18 20:27:02 photonvision avahi-daemon[5632]: Withdrawing address record for fe80::ac7e:36ab:5043:10a1 on eth0.
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2189] device (eth0): state change: ip-config -> failed (reason 'ip-config-unavailable', sys-iface-state: 'managed')
Oct 18 20:27:02 photonvision avahi-daemon[5632]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::ac7e:36ab:5043:10a1.
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2224] manager: NetworkManager state is now DISCONNECTED
Oct 18 20:27:02 photonvision avahi-daemon[5632]: Interface eth0.IPv6 no longer relevant for mDNS.
Oct 18 20:27:02 photonvision NetworkManager[330]: <warn>  [1666139222.2292] device (eth0): Activation: failed for connection 'Wired connection 1'
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2314] device (eth0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2910] dhcp4 (eth0): canceled DHCP transaction
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2911] dhcp4 (eth0): state changed timeout -> terminated
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2952] policy: auto-activating connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2992] device (eth0): Activation: starting connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.2998] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.3018] manager: NetworkManager state is now CONNECTING
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.3030] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.3090] device (eth0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:02 photonvision NetworkManager[330]: <info>  [1666139222.3127] dhcp4 (eth0): activation: beginning transaction (timeout in 45 seconds)
Oct 18 20:27:02 photonvision avahi-daemon[5632]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::ac7e:36ab:5043:10a1.
Oct 18 20:27:02 photonvision avahi-daemon[5632]: New relevant interface eth0.IPv6 for mDNS.
Oct 18 20:27:02 photonvision avahi-daemon[5632]: Registering new address record for fe80::ac7e:36ab:5043:10a1 on eth0.*.
Oct 18 20:27:47 photonvision NetworkManager[330]: <warn>  [1666139267.2155] dhcp4 (eth0): request timed out
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2156] dhcp4 (eth0): state changed unknown -> timeout
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2159] device (eth0): state change: ip-config -> failed (reason 'ip-config-unavailable', sys-iface-state: 'managed')
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2182] manager: NetworkManager state is now DISCONNECTED
Oct 18 20:27:47 photonvision NetworkManager[330]: <warn>  [1666139267.2236] device (eth0): Activation: failed for connection 'Wired connection 1'
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2256] device (eth0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:47 photonvision avahi-daemon[5632]: Withdrawing address record for fe80::ac7e:36ab:5043:10a1 on eth0.
Oct 18 20:27:47 photonvision avahi-daemon[5632]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::ac7e:36ab:5043:10a1.
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2290] dhcp4 (eth0): canceled DHCP transaction
Oct 18 20:27:47 photonvision avahi-daemon[5632]: Interface eth0.IPv6 no longer relevant for mDNS.
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2292] dhcp4 (eth0): state changed timeout -> terminated
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2402] policy: auto-activating connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2491] device (eth0): Activation: starting connection 'Wired connection 1' (356445e8-aaf1-37f8-a5f3-5dd6f8762991)
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2507] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2553] manager: NetworkManager state is now CONNECTING
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2583] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2731] device (eth0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Oct 18 20:27:47 photonvision NetworkManager[330]: <info>  [1666139267.2779] dhcp4 (eth0): activation: beginning transaction (timeout in 45 seconds)
Oct 18 20:27:47 photonvision avahi-daemon[5632]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::ac7e:36ab:5043:10a1.
Oct 18 20:27:47 photonvision avahi-daemon[5632]: New relevant interface eth0.IPv6 for mDNS.
Oct 18 20:27:47 photonvision avahi-daemon[5632]: Registering new address record for fe80::ac7e:36ab:5043:10a1 on eth0.*.
Oct 18 20:28:32 photonvision NetworkManager[330]: <warn>  [1666139312.2255] dhcp4 (eth0): request timed out
Oct 18 20:28:32 photonvision NetworkManager[330]: <info>  [1666139312.2257] dhcp4 (eth0): state changed unknown -> timeout
Oct 18 20:28:32 photonvision NetworkManager[330]: <info>  [1666139312.2260] device (eth0): state change: ip-config -> failed (reason 'ip-config-unavailable', sys-iface-state: 'managed')
Oct 18 20:28:32 photonvision NetworkManager[330]: <info>  [1666139312.2293] manager: NetworkManager state is now DISCONNECTED
[... cut ...]
Oct 18 20:29:17 photonvision NetworkManager[330]: <info>  [1666139357.2812] dhcp4 (eth0): state changed timeout -> terminated
Oct 18 20:29:22 photonvision dhclient[5637]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 14
Oct 18 20:29:36 photonvision dhclient[5637]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 15
Oct 18 20:29:51 photonvision dhclient[5637]: No DHCPOFFERS received.
Oct 18 20:29:51 photonvision dhclient[5637]: No working leases in persistent database - sleeping.
```
Let me know if there's anymore relevant information I can provide.

bznerd, 2022-11-25:
Same behavior on a pi 4.

mdurrani808, 2022-11-27:
Is this on a robot or on a laptop? Have you tried setting a Static ip?

bznerd, 2022-12-03:
Static IP works, mDNS has the described behavior regardless of on or off robot.

mcm001, 2023-01-06:
I'm not familiar enough with mdns stuff to even begin to guess what's up here. I know we run `sudo service avahi-daemon restart` every time photon starts or network settings change, could that be causing issues?

closed 2023-03-04 by bznerd

mcm001, 2023-03-04:
I'm still confused _why_ it went away, but yay