PhotonVision/photonvision #623: Multicamera doesn't work with two of the same cameras

mdurrani808, 2022-12-01:
Fix would be to reference cameras based on their ports rather than their name.

asid61, 2022-12-03:
To elaborate on this, this problem was consistent with AR0144 cameras, which cannot be renamed using the Arducam utility. Would be really nice if we could run multiple of these.

mdurrani808, 2023-01-12:
@drakeerv Due to how thorough this change would be, we are planning on pushing the fix to next season. For now, just use two different cameras. You should do something other than a Lifecam for apriltags in general

fovea1959, 2023-01-24:
not sure this is a big change on Linux. use the /dev/v4l/by-path paths to access cameras....

mcm001, 2024-11-02:
I believe this issue is OBE, but reopen if not!

closed 2024-11-02 by mcm001