PhotonVision/photonvision #631: occasionally disconnecting from rpi

SammyPants37, 2022-12-14:
I'm not sure if I'm meant to be using it this way but I am connected to a rpi running photonvison via ethernet and it will somewhat regularly loose connection to the rasperry pi. I am running 2023 beta 6 on a pi 4

SammyPants37, 2022-12-14:
I am off robot and running dchp (assuming it isn't static with a fresh install)

mdurrani808, 2022-12-14:
Is your Pi plugged into a router or your computer?

mdurrani808, 2022-12-14:
If into a computer, plug it into a router. If its in a router already, check your power supply and send the logs to make sure nothing is crashing.

SammyPants37, 2022-12-14:
I shall try plugging it into the router. now should I connect directly to the router or is the wifi network fine

mcm001, 2022-12-14:
WDYM "stop responding", exactly?

SammyPants37, 2022-12-14:
the camera feed will show a loading gif and it will stop showing targets (I haven't tried out code yet) either giving it time or unplugging the ethernet and plugging back in fixes it I'm not quite sure what.

mdurrani808, 2022-12-14:
Wifi isn't enabled on PhotonVision, plug the Pi directly into the router and then access the webdashboard on a computer on the same network.

SammyPants37, 2022-12-14:
I meant more weather to connect to the router via ethernet or wifi but I'm guessing that doesn't matter too much

SammyPants37, 2022-12-25:
I have moved to a limelight with photonvision that is connected to a robot and I don't think I've ran into connection problems

closed 2022-12-25 by mdurrani808