PhotonVision/photonvision #634: PhotonVision on a Romi always requires that the filesystem be Read/Write (no support for read-only)

mjhealy, 2022-12-16:
## Description
The current [Romi installation docs](https://docs.photonvision.org/en/latest/docs/getting-started/installation/install/romi.html) indicate that the Romi needs to be in a "read/write" configuration during the installation process (e.g., to install dependencies and the JAR file, and to set up the service), and "in order for settings, logs, etc. to be saved / take effect".

However, when running PhotonVision on our Romi units (after updating the installed JAR file to use the RasPi version, per issue #566, and updating the memory allocated to the JVM to 1GB vs. the 512MB default), the software fails to start successfully unless and until the Romi is switched to "read/write" mode, even if it's already been previously configured.

This means that we run the risk of corruption to the SD card image if the Romi is turned off without a safe shutdown, or if the battery power drops too low and the unit suffers a brownout (e.g., if the Romi is left running too long by accident, or during a lengthy debugging session/community demo), which is concerning to say the least.

## Steps to reproduce
1. Install PhotonVision on the Romi.
   * Note: in our experience, to make this work correctly (with our USB camera) we also needed to update the installed JAR file to use the RasPi version, per issue #566, and update the memory allocated to the JVM to 1GB vs. the 512MB default.
2. Put the Romi into "r/w" mode.
3. Configure the camera and pipeline.
   * For us, this included renaming the camera to something meaningful, and configuring the pipeline for the April tags expected in the 2023 FRC game.
   * We did not do extensive configuration, such as camera calibration, etc.
4. Reboot the Romi, so that it comes up in "r/o" mode.
5. Connect to the PhotonVision server from a web client.
   * Confirm that the data for the configuration fails to load correctly, and the camera streams are never available.
6. Disconnect from the PhotonVision server.
7. Reboot the Romi, and put it into "r/w" mode. 
8. Connect to the PhotonVision server from a web client.
   * Confirm that the data for the configuration loads correctly, the camera stream is available, etc.

## Alternate steps to reproduce
1. Execute steps 1-4, as above.
2. Connect to the Romi via SSH.
3. Manually shut down the PhotonVision server (via `sudo systemctl stop photonvision`)
4. Change working directory to `/opt/photonvision`.
5. Verify that the Romi is in "r/o" mode.
6. Manually run the PhotonVision server (`sudo /usr/bin/java -jar /opt/photonvision/photonvision.jar -Xmx1024m`).
7. Connect to the PhotonVision server from a web client.
   * Confirm that the data for the configuration fails to load correctly, and the camera streams are never available.
8. Close the web client.
9. Terminate the PhotonVision server via <Control-C> in the SSH session.
   * A log of the console output from this session is provided in [photonVisionFailedRun.txt](https://github.com/PhotonVision/photonvision/files/10242545/photonVisionFailedRun.txt)
   * Here is a screenshot from the web client session, for review:
[image]
10. Switch the Romi into "r/w" mode.
11. Manually run the PhotonVision server (`sudo /usr/bin/java -jar /opt/photonvision/photonvision.jar -Xmx1024m`).
12. Connect to the PhotonVision server from a web client.
   * Confirm that the data for the configuration **does** load correctly, and the camera stream is available, tracking is performed, etc.
13. Close the web client.
14. Terminate the PhotonVision server via <Control-C> in the SSH session.
   * A log of the console output from this session is provided in
[photonVisionSuccessfulRun.txt](https://github.com/PhotonVision/photonvision/files/10242543/photonVisionSuccessfulRun.txt)
   * Here is a screenshot from the web client session, for review: [image]
15. Switch the Romi into "r/o" mode, and then power off safely.

## Expected behavior
* If the software does require a writeable filesystem at all times in order to run successfully, then the documentation should clearly state this as a hard requirement, rather than as a requirement only for logging/configurations to be _saved_.
* If the software does **not** require a writeable filesystem at all times in order to run successfully, then the software should be updated in order to allow this to happen as intended.

mcm001, 2022-12-16:
OK so my best bet right now is that we aren't extracting the AprilTagJNI to a proper temp directory, and that's breaking things? Does this also happen with a 2022 version of Photon?

mcm001, 2022-12-16:
Wait actually I'm less sure about that -- there's no other way you'd see `Created detector at idx 1` printed. This has me stumped. IIRC 2022 photon has been tested with read-only mode in the past, maybe something new for 2023 caused a regression?

mjhealy, 2022-12-16:
For what it's worth, the code I'm running was built with the 2022 WPILib tools (specifically, the WPILib 2022.4.1 Release), and is running on a Romi with (IIRC) the WPILibPi 2022.1.1 Release image installed.

So any concerns about the changes for 2023 are likely to be a red herring.

mcm001, 2022-12-16:
If you could try PhotonVision v2022.2.0 that would be awesome. I know read only mode was working at one point in the past, I just don't have a romi to do regression testing with

mjhealy, 2022-12-16:
I uploaded PhotonVision v2022.2.0 to the Romi and confirmed that this version allows the software to provide at least a basic image flow through when the Romi is in "r/o" mode.  (It apparently failed to load any of the prior configuration data, including team number, but that's presumably due to the fact that the pipeline configuration was created with the 2023 beta, and included features like the April tags that aren't supported with the 2022 image.)

So, yes, it seems safe to assume that it was a code/data change someplace between the 2022.2.0 release and the current beta for 2023.

If there's anything (short of shipping you one of the team's Romis :-) that I can do to help trouble-shoot this, or to gather information that could lead to a resolution of the issue, I'm happy to do so.  Please let me know!

mcm001, 2022-12-16:
Thanks for testing! My next theory at this point is how we (prior to 2023 beta 7, coming soontm) deal with extracting the apriltag native library. We do it pretty naively, by trying to create a libs folder in the same folder as photon is run from. This will, naturally, fail in read only mode. This is supported by ```[2022-12-15 14:37:16] [VisionModule - AprilTagJNI] [ERROR] IO exception copying apriltag shared object
[2022-12-15 14:37:16] [VisionModule - AprilTagJNI] [ERROR] Failed to load AprilTag Native Library!
```
In beta 7, we'll be extracting the apriltag library the same way wpilib does, so I think that should fix it? Could you give a JAR from here a shot? https://github.com/PhotonVision/photonvision/actions/runs/3716845567
Make sure you choose the correct architecture -- you've probably flashed a 32 bit image and want ` jar-LinuxArm32 `. You can confirm with `uname -m`.

mjhealy, 2022-12-16:
Not overly promising, I fear.  I downloaded the jar-LinuxArm32.zip file, unpacked it, and installed it on the Romi.

After switching back to "ro" mode and running the program, the following output was emitted prior to trying to connect to the web interface:
```
~pi/runProot@photonvision(ro):/opt/photonvision# ~pi/runPhotonVision.sh 
java.io.IOException: Read-only file system
	at java.base/java.io.UnixFileSystem.createFileExclusively(Native Method)
	at java.base/java.io.File.createNewFile(File.java:1024)
	at org.photonvision.common.logging.Logger.addFileAppender(Logger.java:120)
	at org.photonvision.common.logging.Logger.<clinit>(Logger.java:110)
	at org.photonvision.Main.<clinit>(Main.java:58)
Unable to log to file /opt/photonvision/photonvision_config/logs/photonvision-2022-12-15_02-37-16.log
java.nio.file.NoSuchFileException: /root/.wpilib/nativecache/linux/arm32/libopencv_java460.so.77088dd8f72f471b520a94f440ab61a4
	at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
	at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
	at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:215)
	at java.base/java.nio.file.spi.FileSystemProvider.newOutputStream(FileSystemProvider.java:478)
	at java.base/java.nio.file.Files.newOutputStream(Files.java:219)
	at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:114)
	at org.photonvision.common.util.TestUtils.loadLibraries(TestUtils.java:54)
	at org.photonvision.Main.main(Main.java:289)
[2022-12-15 14:37:17] [General - Main] [INFO] Native libraries loaded.
[2022-12-15 14:37:17] [General - Main] [ERROR] Failed to parse command-line options!
[2022-12-15 14:37:17] [General - Main] [INFO] Logging initialized in debug mode.
[2022-12-15 14:37:17] [General - ShellExec] [DEBUG] Executing "cat /proc/device-tree/model"
[2022-12-15 14:37:17] [General - ShellExec] [DEBUG] Got exit code 0
[2022-12-15 14:37:17] [General - Main] [INFO] Starting PhotonVision version dev-v2023.1.1-beta-6-105-gfae6109d on Linux Raspbian (Pi PI_3)
[2022-12-15 14:37:17] [General - ConfigManager] [INFO] Loading settings...
[2022-12-15 14:37:17] [General - ConfigManager] [DEBUG] Making root dir writeable...
[2022-12-15 14:37:17] [General - ConfigManager] [ERROR] Could not make root dir writeable!
[2022-12-15 14:37:17] [General - ConfigManager] [INFO] Hardware config does not exist! Loading defaults
[2022-12-15 14:37:19] [Camera - CameraConfiguration] [DEBUG] Creating camera configuration for UsbCamera USB Camera (AKA RomiCamera) at /dev/video0
Exception in thread "main" java.lang.UnsatisfiedLinkError: org.opencv.core.Mat.n_Mat()J
	at org.opencv.core.Mat.n_Mat(Native Method)
	at org.opencv.core.Mat.<init>(Mat.java:23)
	at org.opencv.core.MatOfPoint3f.<init>(MatOfPoint3f.java:12)
	at org.photonvision.vision.target.TargetModel.listToMat(TargetModel.java:172)
	at org.photonvision.vision.target.TargetModel.<init>(TargetModel.java:152)
	at org.photonvision.vision.target.TargetModel.<clinit>(TargetModel.java:36)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/java.lang.Class.getEnumConstantsShared(Class.java:3554)
	at java.base/java.lang.Class.getEnumConstants(Class.java:3530)
	at com.fasterxml.jackson.databind.util.EnumResolver._enumConstants(EnumResolver.java:181)
	at com.fasterxml.jackson.databind.util.EnumResolver._constructFor(EnumResolver.java:64)
	at com.fasterxml.jackson.databind.util.EnumResolver.constructFor(EnumResolver.java:53)
	at com.fasterxml.jackson.databind.deser.BasicDeserializerFactory.constructEnumResolver(BasicDeserializerFactory.java:2406)
	at com.fasterxml.jackson.databind.deser.BasicDeserializerFactory.createEnumDeserializer(BasicDeserializerFactory.java:1689)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createDeserializer2(DeserializerCache.java:371)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createDeserializer(DeserializerCache.java:349)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCache2(DeserializerCache.java:264)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCacheValueDeserializer(DeserializerCache.java:244)
	at com.fasterxml.jackson.databind.deser.DeserializerCache.findValueDeserializer(DeserializerCache.java:142)
	at com.fasterxml.jackson.databind.DeserializationContext.findNonContextualValueDeserializer(DeserializationContext.java:581)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.resolve(BeanDeserializerBase.java:539)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCache2(DeserializerCache.java:293)
	at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCacheValueDeserializer(DeserializerCache.java:244)
	at com.fasterxml.jackson.databind.deser.DeserializerCache.findValueDeserializer(DeserializerCache.java:142)
	at com.fasterxml.jackson.databind.DeserializationContext.findContextualValueDeserializer(DeserializationContext.java:558)
[... cut ...]
	at org.photonvision.common.configuration.ConfigManager.loadCameraConfigs(ConfigManager.java:322)
	at org.photonvision.common.configuration.ConfigManager.load(ConfigManager.java:186)
	at org.photonvision.Main.main(Main.java:327)
```

Attempts to connect to the service with my web browser are failing, with no indication of an attempted connection in the debugging output.  (Which syncs with the absence of any indication that the app is listening on the port, though the application *is* still running, and did so until I hit <Control-C>.)

mcm001, 2022-12-17:
Pain. `java.nio.file.NoSuchFileException: /root/.wpilib/nativecache/linux/arm32/libopencv_java460.so.77088dd8f72f471b520a94f440ab61a4`. This looks like an error from even within wpilib, somehow?? And it works if you switch to writable?

mjhealy, 2022-12-17:
To be honest, I didn't even try, given that it was falling down even worse than before.  (*wry grin*)

Having done so now, I see the following, which makes it _feel_ as though it's not even getting as far as it did in "ro" mode, though perhaps you're sending the debugging logs to `stdout` in "ro" mode, and to the file when you can do so under "rw" mode?

```
root@photonvision(rw):/opt/photonvision# ~pi/runPhotonVision.sh 
Exception in thread "main" java.lang.UnsatisfiedLinkError: /root/.wpilib/nativecache/linux/arm32/libopencv_java460.so.77088dd8f72f471b520a94f440ab61a4: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /root/.wpilib/nativecache/linux/arm32/libopencv_java460.so.77088dd8f72f471b520a94f440ab61a4)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:121)
	at org.photonvision.common.util.TestUtils.loadLibraries(TestUtils.java:54)
	at org.photonvision.Main.main(Main.java:289)
```

mjhealy, 2022-12-17:
Follow-up thought: if you're unpacking shared libraries into a temp directory on start-up, could part of the issue simply be that I'm not waiting long enough (certainly more than 30sec, and I *think* more than a minute) for the start-up process to complete, and the app to be ready?

If so, would it be possible to add some debugging logs to better indicate the application's progress (e.g., "Unpacking library X, this will be slow", followed some other status when ready)?

mcm001, 2022-12-25:
OOHHHHH actually ok this is good!
> Exception in thread "main" java.lang.UnsatisfiedLinkError: /root/.wpilib/nativecache/linux/arm32/libopencv_java460.so.77088dd8f72f471b520a94f440ab61a4: /lib/arm-linux-gnueabihf/libstdc++.so.6: [...]

This is because the 2023 wpilib is built against C++ 20, but the romi image is probably from 2022. Is there a 2023 romi image you can try with?

mcm001, 2023-01-18:
Should be a new release out now?

mjhealy, 2023-01-18:
Yup.  I'll try to install it this evening or tomorrow (when my day job schedule and other commitments allow :-) and report back.