PhotonVision/photonvision #643: Undistorting 2d detection results

amquake, 2022-12-21:
Lens distortion has a pretty severe effect on the accuracy of the yaw and pitch to the target.
Here is a simulated robot estimating the translation of Apriltags and its own pose with visible, distorted target corners using trigonometry with known camera and target heights (yellow is tag and robot poses estimated from this strategy):

https://user-images.githubusercontent.com/7953350/208889412-552d0fb0-6761-4af1-81b2-79c25cdaba3a.mp4

Here is the same process with the target corners and yaw/pitch undistorted first:

https://user-images.githubusercontent.com/7953350/208889719-4de6a6e3-a381-4bcc-bff5-3ee399a94489.mp4

You can see that undistorting the 2d target info makes a big difference for the accuracy of this estimation strategy.

When a camera is calibrated for the current resolution, the 2d target info can be easily undistorted before being sent to the user. This could be done by default or exposed through a toggle in the UI.

amquake, 2022-12-22:
Not sure if this constitutes a new issue, but on a similar note the FOV could also be automatically determined from a valid calibration instead of the entry in the dashboard settings page for 2d.

Bankst, 2023-01-20:
> Not sure if this constitutes a new issue, but on a similar note the FOV could also be automatically determined from a valid calibration instead of the entry in the dashboard settings page for 2d.

this should be a separate issue but it is closely related

PascalSkylake, 2024-01-29:
> Corners undistorted by #699 -- we still don't undistort pitch/yaw though

I know this is an old issue, but will there ever be an option to undistort pitch/yaw, including targets other than apriltags? I can't speak for everyone, but in our use case it's fairly noticeable and annoying.

amquake, 2024-12-09:
Closed by #1250
(Note that the returned target corners on non-apriltag targets are still distorted)

closed 2024-12-09 by amquake

mcm001, 2024-12-09:
The reported pixel numbers are also still distorted. we should really Make sure we are at least consistent across all target types.