PhotonVision/photonvision #648: Test mode throws exceptions and crashes

prensing, 2022-12-24:
Version a little after v2023.1.1-beta-6, running on LInux Mint (based on Ubuntu 22.04).

I am investigating test mode with Apriltags, but it throws exceptions and frequently crashes. I am building in the code:
   `./gradlew shadowJar`
and then running the jar:
   ` java -jar photon-server/build/libs/photonvision-dev-v2023.1.1-beta-6-26-gb95ce580-linuxx64.jar -t -p ./testimages/`

The problems do seem to change. I think it depends on what has been set in the config tree. Deleting the cameras directory seems to reset it and start the cycle over. 

One of the problems is that the Apriltag detector can return null, and that raises an exception when the values are packed for return:

> [2022-12-23 10:54:40] [VisionModule - VisionRunner - FileFrameProvider2 - PXL_20221111_174026603.png] [ERROR] Exception on loop 493
java.lang.NullPointerException
	at java.base/java.util.List.of(List.java:1032)
	at org.photonvision.vision.pipe.impl.AprilTagDetectionPipe.process(AprilTagDetectionPipe.java:34)
	at org.photonvision.vision.pipe.impl.AprilTagDetectionPipe.process(AprilTagDetectionPipe.java:26)
	at org.photonvision.vision.pipe.CVPipe.run(CVPipe.java:50)
	at org.photonvision.vision.pipeline.AprilTagPipeline.process(AprilTagPipeline.java:143)
	at org.photonvision.vision.pipeline.AprilTagPipeline.process(AprilTagPipeline.java:38)
	at org.photonvision.vision.pipeline.CVPipeline.run(CVPipeline.java:62)
	at org.photonvision.vision.processes.VisionRunner.update(VisionRunner.java:75)
	at java.base/java.lang.Thread.run(Thread.java:829)

Other times, I get various core dumps and JVM crashes:

> [2022-12-23 10:59:29] [VisionModule - VisionSourceSettables] [INFO] Setting video mode to FPS: 30 Width: 640 Height: 480 Pixel Format: kMJPEG
[2022-12-23 10:59:29] [VisionModule - AprilTagDetector] [DEBUG] Creating detector with params AprilTagDetectorParams{tagFamily=tag36h11, decimate=1.0, blur=0.0, threads=1, debug=false, refineEdges=true}
[2022-12-23 10:59:29] [VisionModule - AprilTagDetector] [DEBUG] Creating detector with params AprilTagDetectorParams{tagFamily=tag36h11, decimate=1.0, blur=0.0, threads=1, debug=false, refineEdges=true}
[2022-12-23 10:59:29] [VisionModule - AprilTagDetector] [DEBUG] Creating detector with params AprilTagDetectorParams{tagFamily=tag36h11, decimate=1.0, blur=0.0, threads=1, debug=false, refineEdges=true}
[2022-12-23 10:59:29] [VisionModule - AprilTagDetector] [DEBUG] Creating detector with params AprilTagDetectorParams{tagFamily=tag36h11, decimate=1.0, blur=0.0, threads=1, debug=false, refineEdges=true}
Created detector at idx 3
Created detector at idx 4
Created detector at idx 5
double free or corruption (out)
Aborted (core dumped)

I don't quite remember, but I think the following happens when I try to change the Apriltag family to 16h5:
> Created detector at idx 8
[2022-12-23 11:00:21] [General - ConfigManager] [DEBUG] Saving to disk...
[2022-12-23 11:00:21] [General - ConfigManager] [INFO] Settings saved!
[2022-12-23 11:00:24] [VisionModule - AprilTagDetector] [DEBUG] Creating detector with params AprilTagDetectorParams{tagFamily=tag36h11, decimate=2.0, blur=0.0, threads=2, debug=false, refineEdges=true}
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f30ef24b47e, pid=1569003, tid=1569093
#
# JRE version: OpenJDK Runtime Environment (11.0.17+8) (build 11.0.17+8-post-Ubuntu-1ubuntu222.04)
# Java VM: OpenJDK 64-Bit Server VM (11.0.17+8-post-Ubuntu-1ubuntu222.04, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libc.so.6+0xa547e]  free+0x1e
#
# Core dump will be written. Default location: Core dumps may be processed with "/lib/systemd/systemd-coredump %P %u %g %s %t 9223372036854775808 %h" (or dumping to /scratch/ligerbots/projects/photonvision/core.1569003)
#
# An error report file with more information is saved as:
# /scratch/ligerbots/projects/photonvision/hs_err_pid1569003.log
Created detector at idx 9
#
# If you would like to submit a bug report, please visit:
#   https://bugs.launchpad.net/ubuntu/+source/openjdk-lts
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted (core dumped)

mcm001, 2023-01-04:
Both of these issues should be improved in Beta 8 -- WPILib refactored the apriltag JNI to be better, and we null-check the input to that List.of. Could you try again with beta 8?

prensing, 2023-01-06:
Yes, beta8 looks much better. I don't get any crashes.

closed 2023-01-06 by prensing