PhotonVision/photonvision #663: JAVA backend quit unexpectedly

FUNDAI, 2022-12-29:
After "Settings Saved ! " ,  java backend can't insist for a long time , i can't even key the team number , then just jump out . 
[photonvision-2022-12-29_07-30-25.log](https://github.com/PhotonVision/photonvision/files/10319180/photonvision-2022-12-29_07-30-25.log)

mdurrani808, 2022-12-30:
If you click the settings multiple times in a row, it is possible for the backend to crash. Otherwise, more explanation would be necessary.

FUNDAI, 2023-01-02:
> If you click the settings multiple times in a row, it is possible for the backend to crash. Otherwise, more explanation would be necessary.

I record the situation what I met  , and the red line is
 "[2023-01-02 19:10:44] [General - NetworkTablesManager] [ERROR] [NetworkTablesManager] Could not connect to the robot! Will retry in the background..." 
https://user-images.githubusercontent.com/96116846/210223916-1fc31bf0-ce68-41c8-bb3b-189bcfeb0d80.mp4

gerth2, 2023-01-02:
> Could not connect to the robot! Will retry in the background..."
This should be expected if there is no roboRIO running a Network Tables server.

Unfortunately this one is strange, I haven't seen behavior  like this before. The log shows it's running a one-off dev build from a few releases back. Does the [beta 7](https://github.com/PhotonVision/photonvision/releases/tag/v2023.1.1-beta-7) release have the same behavior?

FUNDAI, 2023-01-03:
> This should be expected if there is no roboRIO running a Network Tables server.
>
> Unfortunately this one is strange, I haven't seen behavior like this before. The log shows it's running a one-off dev [...]

Oh , when I update to beta 7 it fixed but when you enabled your built-in Webcam , it will show the error message 
"[2023-01-03 23:50:15] [VisionModule - VisionRunner - USBFrameProvider - opencv_USB2.0_HD_UVC_WebCam] [ERROR] Exception on loop 588" and crash  , however I think this not a big deal just disabled it and it can work normally 
, and thank you for solving my problem.

mcm001, 2023-01-04:
Sounds like we're good now? Closing the issue but feel free to reopen if you have more trouble

closed 2023-01-04 by mcm001