PhotonVision/photonvision #724: Driver Mode and Pipeline Commands change Network Tables but Not the UI v2023.1.2

icemannie, 2023-01-09:
Using commands to set and reset Driver Mode and change Pipeline index causes the NT values to change but the UI doesn’t.

Changing the same settings in the UI don’t reflect in network tables.

Ornamus, 2023-01-15:
I noticed this today too - and it looks like that while the Network table pipeline index is causing, the actual pipeline itself is not changing.

icemannie, 2023-01-16:
It is essentially unusable in competition without the pipeline feature. The Driver Mode settings can be achieved through a pipeline.

mcm001, 2023-01-18:
Oh that's whacky. Does Glass work?

mcm001, 2023-01-30:
Just confirmed _something_ is weird. I was able to toggle driver mode from Glass and from the web dash, but switching pipeline indexes doesn't seem to trigger the camera to change its video mode...

https://discord.com/channels/725836368059826228/725846784131203222/1069462210335744040

arielCheng218, 2023-02-22:
Please fix this soon if you can! I'm using PhotonVision v2023.4.2 and I'm having the same issue as @icemannie. Switching pipelines with `setPipelineIndex` only changes the pipelineIndex in NetworkTables, but not the web interface. Pipeline type doesn't seem to have anything to do with it too; switching between retroreflective piplines, a retroreflective and an apriltag pipeline, or two apriltag pipelines all don't work.

arielCheng218, 2023-02-22:
I'm using v2023.4.2 on Limelight 2

mcm001, 2023-02-23:
with v2023.4.2 with my Pi here locally running a NT server and robot simulation connected (also on 2023.4.2) I cannot reproduce. I can enable/disaable driver mode and switch pipeline indexes with setPipelineIndex OK.

Gold856, 2025-04-12:
Works fine in 2025.1.1.

closed 2025-04-12 by Gold856

samfreund, 2025-04-12:
> Works fine in 2025.1.1.

Did u mean 2023.3.1?

Gold856, 2025-04-12:
No, I literally tested it just today on 2025.1.1 as well.