PhotonVision/photonvision #727: Photonvision dashboard accessible but camera streams are not

fixermark, 2023-01-10:
Browser: Chrome (up-to-date version)
Platform: Raspberry Pi
Version: [photonvision-v2023.1.2-linuxarm64-image_RaspberryPi.xz](https://github.com/PhotonVision/photonvision/releases/download/v2023.1.2/photonvision-v2023.1.2-linuxarm64-image_RaspberryPi.xz)

Description:
We have successfully imaged PhotonVision to a Pi and are able to connect at http://photonvision.local:5800. 

We navigate to dashboard

Expected result:
- camera feeds are displayed (pre-pipeline and post-pipeline)

Observed result:
- no camera feeds display (yellow boxes show where camera feeds should be, but no camera feeds appear)
- Indicator light on the camera indicates the camera is on.

Popping the Chrome Inspector open shows that it is unable to access photonvision.local:1181 or photonvision.local:1182, receiving ERR_CONNECTION_REFUSED and / or ERR_NAME_NOT_RESOLVED. The dashboard retries repeatedly, but the retries fail in the same way.

fixermark, 2023-01-11:
Closing this issue as unable to reproduce. On another of our Raspberry Pis, after restarting the desktop connected to it after setting its static IP, this worked correctly. I don't have enough additional detail to take a stab at suggesting what went wrong; best suggestion I can give is "If things break while testing, try rebooting the desktop machine."

closed 2023-01-11 by fixermark

EmeraldWither, 2023-01-13:
I have been also having this same problem on a Beelink Mini PC. Rebooting our laptop, and even changing to a different laptop did not seem to fix the problem.