PhotonVision/photonvision #737: PhotonVision camera feed freezes on AprilTag 3d mode enable

superpenguin612, 2023-01-13:
Enabling 3D AprilTag detection for the first time causes camera feed to freeze whenever an AprilTag is in frame (feed comes back when tag is removed from frame), reboot fixes the issue and it doesn't appear again.

Steps to reproduce:
1. install the v2023.1.2-linuxarm64-image_RaspberryPi image on an rpi 4 model b
2. boot up pi/pv, everything is connected via ethernet on a test bench
3. plug in a OV9281 camera
4. set pipeline to apriltags (2d)
5. calibrate camera
6. switch apriltag mode to 3d

mcm001, 2023-01-16:
Just tried on my Pi but couldn't reproduce. Switched to apriltag 2d pipeline at 640x480, calibrated at 960x720, switched to apriltags at 960x720, and enabled 3d.

superpenguin612, 2023-01-16:
Did you have an AprilTag in view, and was it able to render the 3d box around it? The same issue happened again after we reset our Pi due to an unrelated problem.

mcm001, 2023-01-16:
Yes, apriltag in view the whole time. Box was rendered right away for me.

superpenguin612, 2023-01-16:
Interesting, I will try to reproduce again the next time I can.

srimanachanta, 2023-10-28:
@superpenguin612 is this still relevant?

superpenguin612, 2023-10-28:
Not still relevant as far as I can tell, must have been fixed at some point.

closed 2023-10-28 by superpenguin612