PhotonVision/photonvision #755: Feature Request: Add ROI filter

dcowden, 2023-01-20:
Photon Vision is awesome!  Where have you been all my FRC life!?

Using PV for the first time this year, and we love it.  The only issue we've found so far that has us yearning for our CV2 libs is the ability to apply a region of interest ( ROI).  In many years ( this one included), it's possible to dramatically increase pipeline performance and quality  by ignoring large portions of the image.  This year, since the apriltags are all at a known level, a horizontal strip amounting to < 25% of the total image area is needed.

In the offseason, we could submit a PR for this if it would be of interest.

mdurrani808, 2023-01-20:
Agreed that this would be a valuable feature. PR welcome, or someone else may get to it.

mcm001, 2023-01-20:
Pretty easy to add to our Frame provider abstraction tho offsets would need to be added lots of places

mcm001, 2024-11-02:
Superseded by #1347

closed 2024-11-02 by mcm001