PhotonVision/photonvision #766: Support Realtime Recording

carlosgj, 2023-01-27:
It would be helpful to record each frame during a match, in order to diagnose glitches and tweak parameters. 

Obviously, this would be need to be optional, and would only be plausible on coprocessors with bulk storage on a fast connection (USB3, SATA, etc.)

mcm001, 2023-01-30:
The encoding overhead here seems pretty substantial -- worth considering, though

mdurrani808, 2023-01-30:
Maybe when/if H264 streaming becomes a thing within CSCore/PV/

tyrius02, 2024-10-02:
I don't know whether this is relevant or not.
ffmpeg supports use of the H.264 and H.265 encoders baked into the RK3588. Perhaps that would alleviate some of the CPU load, though only for a select set of boards.

mcm001, 2024-10-02:
That's also an intrinsically lossy compression. I'm not sure if I buy that being genuinely useful for debugging if you can't replay the actual inputs to the detector. 

Of course, the only solution here is to go full 971 mode. /s

tyrius02, 2024-10-03:
Huffman YUV (HFYU) is a lossless codec and is supported by ffmpeg and OpenCV. The resulting .avi file will be quite large, of course. I've no idea how much additional load this'd put on the CPU.