PhotonVision/photonvision #772: Native Support for new Camera Module 3 camera

team3572, 2023-01-27:
Running latest dev version of PV (g37a7d378) on Raspberry Pi 4
Attached is new Camera Module 3 NoIR Wide
Does seem to work but Camera shows as "Unknown"
Not sure if this is ok or if PV needs to have some configuration to fully take advantage of this camera. Like it has 120 diagonal FOV.

mcm001, 2023-01-28:
Picam V3 is still not supported yet (given it came out like 2 weeks ago, lol). If you send journalctl logs that could be helpful? Settings tab -> export current logs

team3572, 2023-02-03:
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/10581075/photonvision-journalctl.txt)

Here you go.
I just updated my Pi with latest image and first thing I noticed is the camera image is far out of focus so I'm thinking the camera is at some default focus and there is no way to change it using PV.

mcm001, 2023-02-03:
Yep, that's correct. We don't control focus. Changes will need to be made to photon-libcamera-gl-driver -- there's a PR open to do that right now, actually

mcm001, 2023-02-03:
https://github.com/PhotonVision/photon-libcamera-gl-driver/pull/4 from @realSaddy

realSaddy, 2023-02-03:
I haven't had time to finish the PR up. The driver works for allowing autofocus. See https://github.com/PhotonVision/photonvision/pull/759 for the client-side implementation. Hopefully, I'll get some time this weekend.

If you want to focus your camera in the mean time, you can always SSH in and run the cli commands w/ photonvision disabled.

team3572, 2023-02-03:
@realSaddy Thanks for the tip, I'll try that.
Also if you need some testing with the camera let me know.

mcm001, 2024-11-02:
Any movement here?

jackpittenger, 2024-11-02:
I unfortunately don't have hardware anymore, so someone else will need to pick up development :+1: