PhotonVision/photonvision #774: [Bug] Performance degradation

rwlee, 2023-01-28:
I was working on photon vision with the beelink and the performance gradually degraded to less than 4fps with huge lag and multi second latency. After clearing logs and deleting about 20 calibration images, the performance returned though it's still less than what I was seeing with a fresh install.

mcm001, 2023-01-28:
Does the performance degreation persist across restarts? That's interesting

mdurrani808, 2023-01-28:
What version?

rwlee, 2023-01-29:
Yes, it persists across restarts.

I'm not sure which version, but I had this issue reproduce itself on an orange pi Debian. I set up the orange pi using the debian install scripts linked to on the main website last night -- so I assume I'm on the latest release. I again was able to restore not-quite-original performance by removing logs and the saved calibration images. Next time I run into the issue I'll wipe 1 of those before the other and report back if a specific one resolved the issue.

dcowden, 2023-02-11:
@rwlee We appear to be experiencing this same issue. Can you point me to the location of logs and pics you deleted? I'd like to see if this solves it for us.

Also, to be clear, what we are seeing is that despite pipeline latency of about 25ms, it takes a noticable delay (1 second) before we see a target recognized

mcm001, 2023-02-13:
Latency in what, exactly? NT targeting data, or the MJpeg streamer?

dcowden, 2023-02-13:
Hi, sorry for not being clear there!
We're seeing a delay in the time required to see a target identified.  Here's the situation/setup:

(1) camera running at 320x240-- 90 fps reported capability
(2) actual pipeline latency reported 27ms through the photon api and displayed on smart dashboard
(3) camera feeds not displayed on smart dashboard at all
(4) target coordinates printing on smart dashboard also
(4) rpi 4 with PV 2023.3.0
(5) 4 threads. default settings for most things.
Our test setup is to set the robot facing the camera, with the target acquired, and then start rotating on its axis.  

expected behavior: we lose the target when it goes out of view, but we re-acquire when it comes back around, and we have target data for most of the turn ( limited of course by the viewing angle of the camera, which for is is about 160 degrees )

actual behavior: turning > 1 rotation per second or so, we miss seeing any targeting data at all for the entire turn.  we should have had about 400ms = 20 frames with the target data available, but we didnt.

during this time, i watched htop on the pi. I observed cpu at about 50% on all 4 cpus. no swap usage. memory usage , 1G of 4G avaialble.  Load average 1.9 or so. This doesnt look like memory or cpu pressure. 

pipeline latency was trasmitted during the entire turn, so we are not having a problem getting data at all-- we're just getting a big delay in how long before we have an aquired target.

I wouldnt be surprised if there's something obvious we are missing-- this is our first year using PhotonVision.
If it seems like this is a separate issue, I'm happy to report a new issue, and provide whatever other help i can!

When we saw the behavior above, we had been running most of the morning without a restart. When we restarted the pi to switch camera, we noticed things had improved significantly, with only a restart and no other changes.

rwlee, 2023-02-16:
Sorry for the delay, the files I have been deleting is `/opt/photonvision/photonvision_config/` -- `calibImgs/*` + `logs/*` + `__MACOSX/*` + the same files in `opi arducam`

srimanachanta, 2023-12-01:
Was the root cause of this determined? If not, does this performance degradation still occur on the latest dev image?

mcm001, 2023-12-01:
I don't think we got any closer no. My initial theory was that it was just camera streams being delayed, but it sounds like that wasn't the case? There's been a lot of NT changes since last year so this may be worth revisiting

Gold856, 2025-03-20:
We haven't heard of any reports of this happening to other teams, so closing as stale. If this still occurs, feel free to reopen with details.

closed 2025-03-20 by Gold856