PhotonVision/photonvision #781: Wrong units reported

SauquetAlex, 2023-01-31:
Sometimes after starting PV, the reported data from apriltags is in ft instead of meters. Attached is an example of such an example earlier today:
[image]
[image]

After disconnecting and reconnecting to the robot, the units were fixed.

SauquetAlex, 2023-02-04:
Some update: this seems to occur after calibrating a camera. For ~2-3 minutes after calibration, we get these weird outputs until we reload the page.

Gold856, 2025-03-20:
Can't reproduce. This looks like either a bad calibration or really bad tag ambiguity, or both.

closed 2025-03-20 by Gold856