PhotonVision/photonvision #782: Camera stream doesn't load - Limelight 2023.2.2

StephenDay1, 2023-01-31:
We just put the new image on the limelight and it boots properly, unlike with the previous images.  However, at photonvision.local:5800, the camera stream is just yellow with a loading animation.

We have restarted the limelight several times, and refreshed the page several times as well, nothing changes.

TheOven2, 2023-01-31:
We have the exact same problem.

mcm001, 2023-01-31:
New pi image pulled in libcamera 0.0.3, but we build against 0.0.2, and our old pi image does as well. That's fun. Working on a fix -- you can for now downgrade the libcamera package down to 0.0.2 for a quick fix over ssh

mcm001, 2023-01-31:
Try `sudo apt install libcamera0=0~git20230105+5df5b72c-1`?

mcm001, 2023-02-01:
Or try an offline update using a JAR from https://github.com/PhotonVision/photonvision/actions/runs/4059890146

mcm001, 2023-02-01:
Are either of you able to try the latest development release?

StephenDay1, 2023-02-01:
Development release works. Thank you!

closed 2023-02-01 by mcm001