PhotonVision/photonvision #784: Enhancement request: need to know available pipeline names

fovea1959, 2023-02-01:
when setting up PhotonVision, the students know what the pipeline *names* are. They have to select the pipelines from the roboRIO by *index*. It would be a Good Thing™ if they could set by name from the roboRIO, or know what pipeline names are available.

mcm001, 2023-02-01:
It is technically true that names are guaranteed to be unique... hm

srimanachanta, 2023-02-02:
I assume the main issue with this would be that the current method of changing the index couldn't be changed, an additional NT Topic would need to be created, and a better solution might be the make a read-only topic that just displays the pipeline name.

fovea1959, 2023-02-02:
I agree, using my second suggestion would be best. If I have a list of the pipeline names in order, picking the correct index is easy, and no existing team code would need to be modified (you still pick by index). We are just adding a very static string array to the network table.

I can take this enhancement on. Do we assign issues in this project?