PhotonVision/photonvision #792: Upgrade to v2023.3.0 fails

RJMorriss, 2023-02-06:
Updated the photonvision API to v2023.3.0 in order to get rid of the errors produced we tried to update the photonvision on the Orange Pi.

Attempted to upgrade from v2023.1.2 into v2023.3.0 on an Orange Pi 5 using the commands found at the bottom of the "Other Debian-Based Co-Processor Installation Guide". After moving over the new .jar file and restarting the service, checking the status of the service resulted in the following message:

orangepi@photonvision:~$ sudo systemctl status photonvision.service

● photonvision.service - Service that runs PhotonVision
     Loaded: loaded (/etc/systemd/system/photonvision.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-01-16 01:20:36 CST; 5min ago
   Main PID: 802 (java)
      Tasks: 28 (limit: 9187)
     Memory: 101.1M
        CPU: 3.016s
     CGroup: /system.slice/photonvision.service
             └─802 /usr/bin/java -Xmx512m -jar /opt/photonvision/photonvision.jar

Jan 16 01:20:40 photonvision java[802]: Exception in thread "main" java.lang.UnsatisfiedLinkError: 'int edu.wpi.first.networktables.NetworkTablesJNI.getDefaultInstance()'
Jan 16 01:20:40 photonvision java[802]:         at edu.wpi.first.networktables.NetworkTablesJNI.getDefaultInstance(Native Method)
Jan 16 01:20:40 photonvision java[802]:         at edu.wpi.first.networktables.NetworkTableInstance.getDefault(NetworkTableInstance.java:113)
Jan 16 01:20:40 photonvision java[802]:         at org.photonvision.common.dataflow.networktables.NetworkTablesManager.<init>(NetworkTablesManager.java:37)
Jan 16 01:20:40 photonvision java[802]:         at org.photonvision.common.dataflow.networktables.NetworkTablesManager.getInstance(NetworkTablesManager.java:51)
Jan 16 01:20:40 photonvision java[802]:         at org.photonvision.common.hardware.HardwareManager.<init>(HardwareManager.java:98)
Jan 16 01:20:40 photonvision java[802]:         at org.photonvision.common.hardware.HardwareManager.getInstance(HardwareManager.java:62)
Jan 16 01:20:40 photonvision java[802]:         at org.photonvision.Main.main(Main.java:341)
Jan 16 01:20:42 photonvision java[802]: [2023-01-16 01:20:42] [General - ConfigManager] [DEBUG] Saving to disk...
Jan 16 01:20:42 photonvision java[802]: [2023-01-16 01:20:42] [General - ConfigManager] [INFO] Settings saved!

RJMorriss, 2023-02-06:
Regressed through all version starting from v2023.3.0 until a successful connection was made at v2023.2.1; v2023.2.2 was the first version we encounter this issue.

mcm001, 2023-02-07:
Which jar file were you uploading? Can you send the whole config directory?

brhea, 2023-02-07:
The following jar file was uploaded:

[image]
Photonvision-v2023.3.0-linuxarm64.jar

We found this update when we updated the photonvision API in VScode. We did the online update and then we started getting roboRio messages indicating that the API version did not match the photonvision version on the orange pi. So we updated the orange pi and it crashed. 

Is there a way to revert the API to a version that works with earlier photonvision versions? Perhaps a link could be put in the release notes to a json file that works with that version.

Currently I'm not at the school with the robot but will see if we can post entire config directory this evening.

RJMorriss, 2023-02-07:
I used filezilla to look at orangepi/.config. The directory was empty but had a subdirectory with one file in it as shown. We started from scratch and installed the debian bullseye operating system, then ran the photonvision setup instructions exactly. Initial install worked first time with the initial 2023 release of photonvision.

[image]

[htoprc.txt](https://github.com/PhotonVision/photonvision/files/10679417/htoprc.txt)

RJMorriss, 2023-02-07:
The only thing we did special on install was to follow Arnand's instructions for installing on orangePI from Chief Delphi post. There was only 1 modification to your install.sh file:

# for non-uniform CPUs, like big.LITTLE, you want to select the big cores
# look up the right values for your CPU
# AllowCPUs=4-7

I uncommented AllowCPUs=4-7 and ran the installl

mdurrani808, 2023-02-08:
Can you try a full install script?

brhea, 2023-02-09:
This worked.... we needed to set up a new SD card for our backup orangePI anyway, so we started with installing the OS and went through the entire process running the install.sh script and the API now works with photon vision and photon vision doesn't crash.

mcm001, 2023-02-09:
Sweet. Just to confirm, you're on 2023.3.0 now?

brhea, 2023-02-09:
Correct... both the API and photonvision are on 2023.3.0

mcm001, 2023-02-16:
Closing for now since it sounds like things are working again?

closed 2023-02-16 by mcm001