PhotonVision/photonvision #813: Installing on a Windows 11 Mini PC

zholly15, 2023-02-22:
When trying to run after downloading the Windows jar, we keep getting an exception in main of "java.lang.UnsatisfiedLinkError". 

The full error is as follows:
C:\Users\ADMIN>java -jar "C:\Users\ADMIN\Downloads\photonvision-v2023.4.2-winx64.jar"
Exception in thread "main" java.lang.UnsatisfiedLinkError: C:\Users\ADMIN\.wpilib\nativecache\windows\x86-64\opencv_java460.dll.93feb5fadcb0f27c6b5c730f727195f4: Can't find dependent libraries
        at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
        at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2432)
        at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2489)
        at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2689)
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2619)
        at java.base/java.lang.Runtime.load0(Runtime.java:765)
        at java.base/java.lang.System.load(System.java:1835)
        at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:121)
        at org.photonvision.common.util.TestUtils.loadLibraries(TestUtils.java:55)
        at org.photonvision.Main.main(Main.java:299)

I have verified java 11 is installed:
java version "11.0.18" 2023-01-17 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.18+9-LTS-195)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.18+9-LTS-195, mixed mode)

mcm001, 2023-02-23:
Does an earlier version work?

zholly15, 2023-02-24:
No, this is an error occurring across different versions just on this one machine though.

srimanachanta, 2023-10-28:
duplicate of #775

closed 2023-10-28 by srimanachanta