PhotonVision/photonvision #831: OV9281 camera detected but not loading stream/video in photonvision on Raspberry PI 4B

LucasF6, 2023-03-21:
[Photonvision Problem.zip](https://github.com/PhotonVision/photonvision/files/11024188/Photonvision.Problem.zip)

- In the ZIP file, there are images of the wiring and terminal responses from troubleshooting and config.txt, the log, and settings ZIP.
- https://www.amazon.com/Raspberry-External-Monochrome-Bullseye-libcamera/dp/B09WTP5GZH?th=1
- We have gotten it to work with a USB Microsoft LifeCam. 
- The only command in terminal that gave something unexpected was libcamera-hello
- The ribbon cable is connected correctly
- The robot and backend are connected
- When we go on 10.68.84.11:5800 in the browser, the camera pops up but it doesn't show any video. Sometimes it'll show a yellow square with a rotating thingy, but other times that yellow square won't be there.

BytingBulldogs3539, 2024-01-18:
What version are you running? Have you tried the latest 2024 release?

mcm001, 2025-03-20:
OBE - OV9281s are known working on the latest 2025 release per @Juniormunk 's testing. Reopen if not.

closed 2025-03-20 by mcm001