PhotonVision/photonvision #836: python support in 2024

virtuald, 2023-04-03:
It is expected that Python will be an official FRC language in 2024. Currently the RobotPy project maintains a variety of python wrappers around useful vendor libraries, and we expect to maintain those at least initially until vendors decide they want to provide support themselves. Most vendor libraries have no dependencies other than WPILib, and use fairly vanilla C++, so they're fairly easy to provide support for.

Unfortunately, PhotonVision's vendor library has been broken for RobotPy teams this year because.. well, I'm not 100% sure since I haven't personally done the wrapping, but I think it's because the vendor library now uses OpenCV and doesn't embed it (like the apriltags library does) and because some of your C++ objects use move-only semantics (which is incompatible with pybind11).

Given these issues, I don't really want to spend time on PhotonVision python wrappers in 2024, and I'd rather the PhotonVision project support it instead. It seems to me that if you're building the python wheels as part of your CI process, then you'll be more likely to notice any breakages. Building the python wrappers in-tree instead of depending on a built vendor library should be pretty straightforward, and we're happy to provide some initial assistance to copy the wrapper code over and integrate with your CI. We could also transfer over the existing robotpy-photonvision pypi package to the PhotonVision project, or you could use a new python package name.

CC @auscompgeek 

---

For reference, here are some of the patches and PRs for RobotPy.

Patches for 2023.3.0:

* https://github.com/robotpy/robotpy-photonvision/blob/main/photonvision/src/sources.patch
* https://github.com/robotpy/robotpy-photonvision/blob/main/photonvision/src/headers.patch

PR for 2023.4.x that hasn't been resolved: https://github.com/robotpy/robotpy-photonvision/pull/19/files

mcm001, 2023-12-08:
I would be happy to see this mainlined if someone can step up and take ownership of this refactor

mcm001, 2024-01-04:
Closed by https://github.com/PhotonVision/photonvision/pull/1040