PhotonVision/photonvision #837: MultiPNP will run on single tag of one tag id is ninvalid

stephenjust, 2023-04-04:
We've been seeing lots of "phantom" targets from photon vision. While troubleshooting random position jumps, I noted that in the MultiPNP pose estimation strategy (Java), there's a capability to fall back to another strategy if only one tag is visible. However, if two tags are visible and one is rejected as invalid, the MultiPNP logic will run against the single tag.

https://github.com/PhotonVision/photonvision/blob/master/photon-lib/src/main/java/org/photonvision/PhotonPoseEstimator.java#L341

It's not super obvious to me that this is the cause of our pose jumpiness, or if it's a problem at all, but it seemed unintentional given that there is fallback code for a single tag.

stephenjust, 2023-04-17:
> Closed by #817 I think

I don't think so. You're still running the MultiPNP algorithm on a single tag if enough corners are rejected.

srimanachanta, 2023-04-19:
> Closed by #817 I think

To fix this, you need to reject targets before the if statement checking how many targets there are.

mcm001, 2023-04-19:
I am? ```java
        for (var target : result.getTargets()) {
            // Check if tag actually exists in the field layout
            var tagPoseOpt = fieldTags.getTagPose(target.getFiducialId());
            if (tagPoseOpt.isEmpty()) {
                reportFiducialPoseError(target.getFiducialId());
                continue;
            }

            // Now that we know tag is valid, add detected corners
            visCorners.addAll(target.getDetectedCorners());

            // actual layout poses of visible tags -- not exposed, so have to recreate
            var tagPose = tagPoseOpt.get();
            knownVisTags.add(new AprilTag(target.getFiducialId(), tagPose));
        }

        if (result.getTargets().size() < 2 || knownVisTags.size() < 2) {
            // Run fallback strategy instead
            return update(result, this.multiTagFallbackStrategy);
        }

        var cameraMatrixOpt = camera.getCameraMatrix();
        var distCoeffsOpt = camera.getDistCoeffs();
        boolean hasCalibData = cameraMatrixOpt.isPresent() && distCoeffsOpt.isPresent();

        // multi-target solvePNP
        if (hasCalibData) {
            var cameraMatrix = cameraMatrixOpt.get();
            var distCoeffs = distCoeffsOpt.get();
            var pnpResults =
                    VisionEstimation.estimateCamPoseSqpnp(cameraMatrix, distCoeffs, visCorners, knownVisTags);
            var best =
                    new Pose3d()
                            .plus(pnpResults.best) // field-to-camera
                            .plus(robotToCamera.inverse()); // field-to-robot
            // var alt = new Pose3d()
            // .plus(pnpResults.alt) // field-to-camera
            // .plus(robotToCamera.inverse()); // field-to-robot

            return Optional.of(
                    new EstimatedRobotPose(best, result.getTimestampSeconds(), result.getTargets()));
        } else {
            // TODO fallback strategy? Should we just always do solvePNP?
            return Optional.empty();
        }```

stephenjust, 2023-04-19:
Ah yes, that's right, you're checking again and falling back. That should be fine.

commit f6756bdb (2023-06-18): Don't add bad corners to multi-tag solution (#817)

Closes #837

---------
closed 2023-06-18 by mcm001