PhotonVision/photonvision #842: Create build for Limelight 3

srimanachanta, 2023-04-13:
Currently, users can use a Limelight 3 using the following steps. A build should be generated along with the rest for Limelight.

via @mcm001 
steps
1. flash our stock Pi image (NOT limelight)
2. connect to the internet
3. run `sudo wget https://datasheets.raspberrypi.org/cmio/dt-blob-cam1.bin -O /boot/dt-blob.bin`
4. edit `/boot/config.txt` and uncomment `dtoverlay=ov5647`

marshallmassengill, 2023-09-19:
So I finally managed to get this working with the LL3 and the standard rpi 32bit image.  I could not get it working with the 64 bit version yet (I need to retry it now).  General process was:

1. Using the https://github.com/raspberrypi/usbboot tool 
2. Putting the device into a mode where I could use the Raspberry Pi Imager tool to send it a custom image with ssh enabled and password/locale set
3. Connect it to an internet connection
4. run `sudo wget https://datasheets.raspberrypi.org/cmio/dt-blob-cam1.bin -O /boot/dt-blob.bin`
5. edit /boot/config.txt and add `dtoverlay=ov5647`
6. The camera will show up as a libcamera and not a legacy camera

This definitely needs more testing.

mcm001, 2023-09-19:
With a 32 bit OS, I think you would need to recompile our libcamera driver too, right?

srimanachanta, 2023-09-23:
We should add something to docs at the minimum for next season about this

marshallmassengill, 2023-10-12:
I managed to get this working with the 64 bit pi OS finally.  It's definitely just as described - however I could not get this working with the legacy camera stack, only libcamera.  Also of note, the LL3 will take the OV9281 global shutter cameras sold by waveshare - just modify the config.txt to use it.

mcm001, 2023-10-13:
> I managed to get this working with the 64 bit pi OS finally.  It's definitely just as described - however I could not get this working with the legacy camera stack, only libcamera.  Also of note, [...]

Awesome! Yeah, I didn't have success with the legacy camera stack either.

If you have the bandwidth to either write up steps or PR a generator script to our pi-gen repo that would be much appreciated:)

marshallmassengill, 2023-10-13:
It's going to take a bit but we are going to write up something and I'll link it here.  Gotta say, the camera stack for the pi is a complete mess and I'm hoping it gets better with libcamera now.

mcm001, 2023-12-29:
> It's going to take a bit but we are going to write up something and I'll link it here. Gotta say, the camera stack for the pi is a complete mess and I'm hoping it gets better with libcamera now.

@marshallmassengill any chance you’ll get around to this before kickoff? Or was it literally just flash rpiOS and download the new dt-blob.bin?

marshallmassengill, 2023-12-29:
That was pretty much it. We did actually publish the images we made here: https://www.chiefdelphi.com/t/the-zebracorns-behind-the-stripes-design-code-and-build-blog-2023-2024/440094/35?u=marshall

mdurrani808, 2024-02-01:
Closed by #1166.

closed 2024-02-01 by mdurrani808