PhotonVision/photonvision #880: Ask to change hostname instead of just doing it on initial startup

JonathanLindsey, 2023-06-27:
## Description

When I run the jar, it changes my PC's hostname to `photonvision` without asking. This is particularly annoying when running PhotonVision from my development PC for testing.

## Expected behavior

When I run the jar, it should read the current hostname of the PC it is running on and populate the web UI accordingly.

OS images should be pre-configured with the `photonvision` hostname.

If the hostname is not `photonvision` on startup, perhaps a good compromise would be asking if it may change the hostname to `photonvision` at the same time it asks you to set a team number. If the user accepts this change, redirect to the new hostname.

## Platform I experienced the issue on

* Pop_OS Linux (Ubuntu derivative) 
* Using `photonvision-v2023.4.2-linuxx64.jar`

mcm001, 2023-06-27:
How can a user say yes or no if the host name of the device is initially unknown?

We had a disable networking flag at one point you could pass, but I'm not positive if it still works. Maybe that's a better solution for development?

JonathanLindsey, 2023-06-27:
Yes, a bit of a catch 22 but if a user is running the jar, I figure they're already on the machine and either they could lookup the hostname (the command is `hostname` on linux) or the jar could print it out for them on the terminal. On linux they can also just look at they're terminal prompt after the '@' sign.

A command line flag would be better than nothing. If a user doesn't know about the flag or forgets to use it once, it still feels like it's making an administrative action that was wasn't asked for.

mcm001, 2023-06-27:
I think as long as it's called out sufficiently in docs, it's fine? Our main target devices are something we expect to manage, and I don't know of a good way to tell if the program is running from a service or not.

It looks like that config flag to disable networking doesn't exist -- I'm open to a pull request adding it, though! Or making our code actually respect the `shouldManage` field in NetworkConfig.

samfreund, 2025-04-14:
This was added in #1118

TianHengZhuang, 2026-09-30:
Picked this one up: #2638.

`docs/source/docs/advanced-installation/sw_install/advanced-cmd.md` only covered `java -jar`, so `-n` was undocumented. The PR adds a section for it, including the fact that `NetworkManager.initialize` computes `isManaged = shouldManage && !networkingIsDisabled`, so the hostname rewrite, the static IP config and the interface monitoring all get skipped, and the dashboard networking controls are disabled too. It also notes the dashboard is still on 5800 and that the same skip happens when you are not on Linux / not root / no `nmcli`.

Docs only. Let me know if `advanced-cmd.md` is the wrong page for it.

closed 2026-09-30 by gerth2