PhotonVision/photonvision #910: Make tag width configurable

rzblue, 2023-09-14:
Currently tag width is hardcoded to 6". This should be made configurable.

mcm001, 2023-09-19:
This is actually somewhat difficult to do in our code. Can we instead do 2023_6in and 2024_6.5in models?

rzblue, 2023-09-21:
I suppose that would be fine. Is this because of the need to have the target model to draw on the image?

mcm001, 2023-09-22:
Yeah, and right now, target model is saved as an enum. Having a "custom" thing isn't possible to encode in an enum really if we just send the ordinal value/name to the frontend either.

It's a dumb requirement to make everything an enum instance, though. I would be open to just Not Doing That

rzblue, 2023-09-22:
Right now the tag width is a hardcoded value in `AprilTagPipeline` (https://github.com/photonvision/photonvision/blob/master/photon-core/src/main/java/org/photonvision/vision/pipeline/AprilTagPipeline.java#L72)

I made some quick and dirty changes to add the tag width to the apriltag settings, haven't tested it yet though. Of course, the boxes drawn on the image will be incorrect.

closed 2025-03-20 by mcm001