PhotonVision/photonvision #935: Bump gradle version with WPILib, enable windows arm

mcm001, 2023-10-08:
Copy the diff from here: https://github.com/wpilibsuite/allwpilib/pull/5727/files

apoblock, 2026-11-04:
Updating this with a concrete 2027 robot-code failure, now that WPILib ships an official Windows ARM64 installer.

On Windows 11 ARM64 (Qualcomm Snapdragon, `os.arch=aarch64`) with WPILib 2027.0.0-alpha-8, the PhotonLib vendordep still has no `windowsarm64` JNI. The copy we have is `dev-v2027.0.0-alpha-2-89-g4b8d9b50`. Its `photontargeting-cpp` and `photonlib-cpp` entries set `skipInvalidPlatforms: true` and list only:

- `windowsx86-64`
- `linuxsystemcore`
- `linuxx86-64`
- `osxuniversal`

Gradle therefore does not extract Photon natives into `build/jni/release`. WPILib's own sim GUI does load (`halsim_gui.dll` is ARM64). A robot program that constructs a `PhotonCamera` cannot get past native load on this JVM. Our project dies first inside REVLib (`REVLibDriver could not be loaded ... attempted to load for platform windows-arm64`); Photon is the next native vendor after that, via `Vision` / `PhotonCamera`.

This is the desktop vendordep used by robot simulation, separate from the Linux coprocessor images. The closed request in #1621 asked for published Windows ARM64 binaries and was left as "need more evidence of interest." The 2027 alpha installer makes that the default WPILib desktop platform on Snapdragon laptops.

Could `windowsarm64` be added to the published robot vendordep (`photontargeting-cpp` / `photonlib-cpp`) for the 2027 line? An ARM64 JVM cannot load the existing `windowsx86-64` DLLs.

samfreund, 2026-11-04:
> Updating this with a concrete 2027 robot-code failure, now that WPILib ships an official Windows ARM64 installer.
>
> On Windows 11 ARM64 (Qualcomm Snapdragon, `os.arch=aarch64`) with WPILib [...]

Please rewrite without AI.

apoblock, 2026-11-04:
Here's the non-AI brief: PhotonVision is not publishing `windowsarm64` vendordep for WPILib 2027 alpha 8. See comment above for the details.