PhotonVision/photonvision #1013: Add pose standard deviation to Web UI

asid61, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
When I'm testing different camera configurations like exposure, brightness, threads, and iterations, I don't have an easy way of comparing the accuracy of two configs. This means I need a full control system setup and various logging steps to get the pose standard deviation from a camera.

**Describe the solution you'd like**
It would be nice to show the standard deviation of the pose (or multitag pose) over the last 5-10 seconds under or next to where the poses are shown. Sigma X, Y, and Theta(s) would be handy.

**Describe alternatives you've considered**
Running a control system, getting pose over NT, logging the pose, exporting the log, clearing out bad data, and using Excel/Sheets to find the pose standard deviation.

mcm001, 2023-12-20:
Our code doesn't account for angle wrapping when calculating standard devs. Also weird barf in the console per Nolan -- also the targets pane will go poof sometimes

[image]

srimanachanta, 2023-12-20:
> Our code doesn't account for angle wrapping when calculating standard devs. Also weird barf in the console per Nolan -- also the targets pane will go poof sometimes
> 
> [image]

the barf is due to #1030 

Neel to add an AngleModulus method but I'm not sure why because Rotation3d should already handle that

mcm001, 2023-12-20:
Fixed locally PR incoming

PR #1061 (2023-12-28): Stdev-UI: Handle angle wrapping
Closes #1013
closed 2023-12-28 by srimanachanta