PhotonVision/photonvision #1016: Use protobuf for setting serialization

mcm001, 2023-11-22:
JSON is nice and all, but it can struggle with weird things like serializing arrays of polymorphic types. I do hacky things to get around this in my SQL code. If our settings database was instead one big protobuf, we should be able to reduce the quantity of hacky stuff we do. This has the fun side effect of enabling us to support changing pipeline settings over NT (hello, ros services but worse!)

srimanachanta, 2023-11-22:
I'm not a big fan of sending full settings over NT and allowing the rio to control those becasue:
1. It could become a race condition with both the UI and NT manipulating the same settings simultaneously
2. There is no major use case for the rio needing to manipulate settings greater than driver mode and pipeline index. Everything else would require the dashboard cause you need to actually see the results.

mcm001, 2023-11-22:
Yeah fair. Still think protobufs for internal only settings storage are good, though the value prop is basically just "it's better lol"

srimanachanta, 2023-11-23:
> Yeah fair. Still think protobufs for internal only settings storage are good, though the value prop is basically just "it's better lol"

I agree with that, SerDe with Protobuf beats Jackson for me