PhotonVision/photonvision #1028: Shows incorrect memory utilization on PhotonVision Settings

AlphaPranav9102, 2023-11-26:
**Describe the bug**
On the PhotonVision settings page, a statistic is displayed with the CPU Memory Usage. The data displayed here actually displays the free CPU Memory on the device instead of the used CPU Memory. In the images shown below, when SSHing into the Pi, the actual utilized memory is 576 MB but the number shown in Photonvision is 7209 MB which would actually represent the free memory on the device.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the PhotonVision Dashboard
2. Click on the settings tab
3. Scroll down to Hardware Metrics table
4. See error in column 3 labeled "CPU Memory Usage"

**Screenshots / Videos**
[image]
[image]

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): OrangePi 5B
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Running locally on a Pi connected to home ethernet connection with no robot. Accessing PhotonVision dashboard over wifi.
 - PhotonVision Version: v2024.1.1-beta-3
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome 119.0.6045.159
 - Camera(s) Used: Arducam OV 2311 USB Camera

**Additional context**
There is no overclocking on this OrangePi.

mcm001, 2023-11-26:
We should fix the backend to actually report free memory not used

mcm001, 2023-11-27:
Starting points: https://github.com/PhotonVision/photonvision/blob/master/photon-core/src/main/java/org/photonvision/common/hardware/metrics/cmds/LinuxCmds.java and https://github.com/PhotonVision/photonvision/blob/16f63e4d90f0946d96ae4ce924c7748b877124cb/photon-core/src/main/java/org/photonvision/common/hardware/metrics/MetricsManager.java#L115

closed 2024-06-07 by mcm001