PhotonVision/photonvision #1030: Deferred store accessors don't correctly update on settings changes

srimanachanta, 2023-11-28:
When using deferred store accessors, the values within the accessor arent correctly updating which settings are changed on another deferred store. This is best seen in the multitag + targets tab relationship where changes in one deferred store arent reflected in the other. The backend is correctly receiving the requested change so this is a UI issue.

mcm001, 2024-01-01:
This is the same issues as https://github.com/PhotonVision/photonvision/issues/1071 right?

srimanachanta, 2024-01-01:
> This is the same issues as https://github.com/PhotonVision/photonvision/issues/1071 right?

I don't believe so, deferred store access is a camera settings store issue. Might be cause the buffer isn't saved into local storage, maybe we wanna add that.

PR #1104 (2024-01-02): Improve UI stability, reliability, and readability
closes #1090
closes #1030 

Also fixes various styling issues and overflow issues for mobile support
commit e4f475a2 (2024-01-02): Improve UI stability, reliability, and readability (#1104)

closes #1090
closes #1030

Also fixes various styling issues and overflow issues for mobile support
closed 2024-01-02 by mcm001