PhotonVision/photonvision #1035: Swapping CSI cameras results in the old config file being used.

BytingBulldogs3539, 2023-12-08:
**Describe the bug**
Swapping CSI camera module on Raspberry Pi does not perform checks to ensure that the camera is the same model and uses wrong config file for the CSI camera. PhotonVision should generate a new config file for this new camera since it is a different model than the last camera(s).

**To Reproduce**
Steps to reproduce the behavior:
1. Boot photon with one camera plugged into CSI (in my case ov5647)
2. Unplug power
3. Swap CSI camera to another type (in my case IMX708)
4. See the IMX708 show up as OV5647.

**Logs**
[photonvision-journalctl (15).txt](https://github.com/PhotonVision/photonvision/files/13607762/photonvision-journalctl.15.txt)

**Platform:**
 - Raspberry PI 5
 - Camera(s) Used: OV5647(pi cam 1) then IMX708 (pi cam 3)

BytingBulldogs3539, 2023-12-08:
> The way the camera matching code is currently setup, it checks if cameras and configs can be matched by path-id (camera id), then normal path (aka port), and finally by name. So if you plug in a different camera with the same name, it will assume it's the same camera and the ID and port have changed (great for swapping out a camera with the same model). Maybe it's time to revisit that logic? Can't think of a good way to do that without breaking windows compatibility.

I am fairly sure this issue is limited to CSI cameras on the pi 5 (maybe also other pi versions) the device name is RPI-CFE for every camera that is plugged into the CSI port. The device nickname is set to the device model for the case of CSI cameras however the nickname can be changed so we can't check camera model vs nickname. I think we may need to add a camera model string to the camera configuration and check against that.

MrRedness, 2023-12-09:
Ah I see. Should be relatively simple to fix then

BytingBulldogs3539, 2023-12-25:
#1068 Fixes Issue

BytingBulldogs3539, 2023-12-25:
#1068 fixes issue

closed 2023-12-25 by BytingBulldogs3539