PhotonVision/photonvision #1046: Multi-camera support doesn't migrate existing settings database to new format.

crschardt, 2023-12-16:
**Describe the bug**
#1015 added a new column to the "camera" table in the settings database, but it doesn't migrate an existing database to the new format. If there is an existing database, then database access fails with the following error message (h/t quicksinger21 on discord):

```
Dec 17 04:31:56 photonvision java[2280]: [2023-12-17 04:31:56] [General - ConfigManager] [DEBUG] Saving to disk...
Dec 17 04:31:56 photonvision java[2280]: [2023-12-17 04:31:56] [Config - SqlConfigProvider] [DEBUG] Saving to disk
Dec 17 04:31:56 photonvision java[2280]: [2023-12-17 04:31:56] [Config - SqlConfigProvider] [ERROR] Err saving cameras: [SQLITE_ERROR] SQL error or missing database (table cameras has no column named otherpaths_json)
Dec 17 04:31:56 photonvision java[2280]: [2023-12-17 04:31:56] [Config - SqlConfigProvider] [ERROR] org.sqlite.SQLiteException: [SQLITE_ERROR] SQL error or missing database (table cameras has no column named otherpaths_json)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.DB.newSQLException(DB.java:1179)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.DB.newSQLException(DB.java:1190)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.DB.throwex(DB.java:1150)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.NativeDB.prepare_utf8(Native Method)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.NativeDB.prepare(NativeDB.java:126)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.DB.prepare(DB.java:264)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.core.CorePreparedStatement.<init>(CorePreparedStatement.java:46)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.jdbc3.JDBC3PreparedStatement.<init>(JDBC3PreparedStatement.java:31)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.jdbc4.JDBC4PreparedStatement.<init>(JDBC4PreparedStatement.java:25)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.jdbc4.JDBC4Connection.prepareStatement(JDBC4Connection.java:34)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.jdbc3.JDBC3Connection.prepareStatement(JDBC3Connection.java:226)
Dec 17 04:31:56 photonvision java[2280]:         at org.sqlite.jdbc3.JDBC3Connection.prepareStatement(JDBC3Connection.java:206)
Dec 17 04:31:56 photonvision java[2280]:         at org.photonvision.common.configuration.SqlConfigProvider.saveCameras(SqlConfigProvider.java:304)
Dec 17 04:31:56 photonvision java[2280]:         at org.photonvision.common.configuration.SqlConfigProvider.saveToDisk(SqlConfigProvider.java:183)
Dec 17 04:31:56 photonvision java[2280]:         at org.photonvision.common.configuration.ConfigManager.saveToDisk(ConfigManager.java:279)
Dec 17 04:31:56 photonvision java[2280]:         at org.photonvision.common.configuration.ConfigManager.saveAndWriteTask(ConfigManager.java:288)
Dec 17 04:31:56 photonvision java[2280]:         at java.base/java.lang.Thread.run(Thread.java:829)
Dec 17 04:31:56 photonvision java[2280]: 
Dec 17 04:31:56 photonvision java[2280]: [2023-12-17 04:31:56] [Config - SqlConfigProvider] [INFO] Settings saved!
Dec 17 04:33:33 photonvision java[2280]: [2023-12-17 04:33:33] [General - ShellExec] [DEBUG] Executing "journalctl -u photonvision.service > /tmp/photonvision-journalctl12219735344239381215.txt"
```

crschardt, 2023-12-17:
I figured out how to add the column to a table if it's missing. Now I'm working on figuring out how to prevent it from duplicating the camera entry when it does this. I'll try to have a PR soon.

mcm001, 2023-12-17:
Also dumb question what happens if users go backwards? From a version with this new DB format to one before it was changed I mean. Not that we should be promising backwards compatibility during beta, but for future reference

crschardt, 2023-12-17:
That is a great question! I thought maybe it would work and the older version would just ignore the extra column, but testing it lead me to #1053, which makes me think that it could be a lot of work to allow for backward compatibilty.

commit f37a0d03 (2024-01-06): Add database migrations (fixes #1046)  (#1065)

Add database version pragma to SQL database to automatically migrate between versions
closed 2024-01-06 by mcm001