PhotonVision/photonvision #1051: VideoExceptions reported in log file

kinahawi, 2023-12-17:
**Describe the bug**
VideoExceptions are reported in the log file when attempting to set auto exposure mode values. Running PhotonVision on Beelink mini PC with two cameras installed; an Arducam ov9281 and a Global Shutter Camera.
[photonvision-journalctl (1).txt](https://github.com/PhotonVision/photonvision/files/13696717/photonvision-journalctl.1.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot photon vision and look in the log file.

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Beelink Mini-PC running Ubuntu 22.04.3 LTS (jammy):
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Hardwired to local network switch using CHCP
 - PhotonVision Version: 2024 beta 3.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.):
 - Camera(s) Used: Arducam ov9281 and a Global Shutter Camera

**Additional context**
Add any other context about the problem here.

mcm001, 2023-12-17:
That seems to be a (continuing) bug with cscore and property name changes in v4l. Can you try v4l2-ctl --list-controls (I think) and send the output?

kinahawi, 2023-12-17:
Here is the output

***@***.***:~$ sudo v4l2-ctl --list-ctrls

User Controls

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=66 value=66
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=115 value=115
                           gain 0x00980913 (int)    : min=16 max=255 step=1 default=32 value=32
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1 (50 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=0 value=0
         backlight_compensation 0x0098091c (int)    : min=28 max=201 step=1 default=55 value=55

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=8188 step=1 default=156 value=156 flags=inactive
                   pan_absolute 0x009a0908 (int)    : min=-648000 max=648000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-648000 max=648000 step=3600 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=0 max=60 step=1 default=0 value=0

mcm001, 2023-12-18:
WPILib expects these property names:

```
static constexpr char const* kPropExAuto = "exposure_auto";
static constexpr char const* kPropExValue = "exposure_absolute";
```
https://github.com/wpilibsuite/allwpilib/blob/a004c9e05f08c1bac7989582e930bf416fb285c9/cscore/src/main/native/linux/UsbCameraImpl.cpp#L54

Looks like the property names reported by your camera don't even sort of match up. You should patch UsbCameraSource to special case for if the camera in question has different property names.

mcm001, 2023-12-18:
Or ping Peter about https://github.com/wpilibsuite/allwpilib/issues/5478 -- known issue upstream