PhotonVision/photonvision #1052: No wired connections reported incorrectly

kinahawi, 2023-12-17:
**Describe the bug**
I'm getting a message reported that there are no wired connections detected. Clearly there is a wired connect since I'm able to connect to the device using the wired connection.  (see screen shot for error message)
[image]

I'm running Ubuntu 22.04 on a Beelink computer hosting Photonvision.
When I execute ifconfig on the device, it reports an interface named "enp1s0". The network confif.json file had an interface named "eth0". I updated the network config to use the reported interface name but this had no effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Beelink mini-PC running Ubuntu 22.04.3 LTS
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch):
 - PhotonVision Version: 2024 beta 3.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Arducam ov 9281 and Global Shutter Cam

**Additional context**
Add any other context about the problem here.

mcm001, 2023-12-17:
You can see how we determine this on line 96 here. Try executing that command and sending the output here?

https://github.com/PhotonVision/photonvision/blob/master/photon-core/src/main/java/org/photonvision/common/networking/NetworkUtils.java

kinahawi, 2023-12-17:
Here are the results of executing that line:

***@***.***:~$ nmcli -t -f GENERAL.CONNECTION,GENERAL.DEVICE,GENERAL.TYPE device show
GENERAL.CONNECTION:
GENERAL.DEVICE:wlp2s0
GENERAL.TYPE:wifi

GENERAL.CONNECTION:
GENERAL.DEVICE:enp1s0
GENERAL.TYPE:ethernet

GENERAL.CONNECTION:
GENERAL.DEVICE:lo
GENERAL.TYPE:loopback

GENERAL.CONNECTION:
GENERAL.DEVICE:p2p-dev-wlp2s0
GENERAL.TYPE:wifi-p2p

Let me know what else I can provide.

Bill

mcm001, 2023-12-18:
And what about the output of "nmcli conn show"? Looks to me like GENERAL.CONNECTION is empty which we interpret as not being connected

kinahawi, 2023-12-19:
Hi Matt,

The "nmcli connection show" commend just results in a blank line.

"nmcli d" results in :

DEVICE          TYPE      STATE         CONNECTION 
wlp2s0          wifi      disconnected  --         
enp1s0          ethernet  unmanaged     --         
lo              loopback  unmanaged     --         
p2p-dev-wlp2s0  wifi-p2p  unmanaged     --      

So it looks like the network manager is not managing those interfaces? I have little/no experience in setting up networks in Ubuntu. BTW - this machine is running Ubuntu Server.

I tried following some instructions found on “Ask Ubuntu” (https://askubuntu.com/questions/1402636/nmcli-not-showing-connection?newreg=a6d14d73508046c8b7f8b035eb5970ac), but this resulted in a two minute delay on booting the machine. 

Any advice on how to set up the connections so that the network manager .

Thanks for your help!

Bill

mcm001, 2023-12-20:
Here's results from my laptop running a fresh install of Ubuntu 22.04. Not sure why your ubuntu install doesn't default to networkmanager, but it works locally so not sure what gives.

```
matt@photonvision:~$ nmcli connection show
NAME                             UUID                                  TYPE      DEVICE  
Wired connection 1               cc9cb740-6a01-3d6e-ae1a-2311d8feefdf  ethernet  eno1    
docker0                          d140ad16-f518-4164-804a-59db8c2be921  bridge    docker0 
flyfi.com                        fc2c50c7-d7ac-4a4b-a9d9-9efa89496e41  wifi      --      
Hilton Honors                    39afeb11-76aa-4582-8448-bbcdaeee1d92  wifi      --      
Matt's Marvelous 2.4ghz Mansion  5cb99f5d-770d-42bc-bb07-34f5062e1847  wifi      --      
#SFO FREE WIFI                   114d8c2f-615f-4c78-8cdf-2fc762dd7a20  wifi      --      
SpectrumSetup-51                 c94eada9-d65b-4228-898d-9fcbd150fe49  wifi      --      
Wired connection 2               a8fa38a2-8731-3b87-b615-81def8c92dac  ethernet  --      
matt@photonvision:~$ nmcli d
DEVICE        TYPE      STATE                   CONNECTION         
eno1          ethernet  connected               Wired connection 1 
docker0       bridge    connected (externally)  docker0            
wlo1          wifi      disconnected            --                 
p2p-dev-wlo1  wifi-p2p  disconnected            --                 
lo            loopback  unmanaged               --                 
matt@photonvision:~$ nmcli -t -f GENERAL.CONNECTION,GENERAL.DEVICE,GENERAL.TYPE device show
GENERAL.CONNECTION:Wired connection 1
GENERAL.DEVICE:eno1
GENERAL.TYPE:ethernet

GENERAL.CONNECTION:docker0
GENERAL.DEVICE:docker0
GENERAL.TYPE:bridge

GENERAL.CONNECTION:
GENERAL.DEVICE:wlo1
GENERAL.TYPE:wifi

GENERAL.CONNECTION:
GENERAL.DEVICE:p2p-dev-wlo1
GENERAL.TYPE:wifi-p2p

GENERAL.CONNECTION:
GENERAL.DEVICE:lo
GENERAL.TYPE:loopback
matt@photonvision:~$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
matt@photonvision:~$ 
```

kinahawi, 2023-12-21:
Are you running Ubuntu server? I’m running the server version. I was under the impression that NetworkManager is a GUI tool that doesn’t necessarily make sense in a server environment. When I start the NewtorkManager, it adds 2 minutes to the boot process, but then PhotonVision foes see the connection.Sent from my iPhone

mcm001, 2023-12-21:
Looks like your new line formatting got messed up? Not sure how that happened, maybe an email thing?

kinahawi, 2023-12-21:
I guess because I did a reply from my phone.

Are you running Ubuntu server? I’m running the server version. I was under the impression that NetworkManager is a GUI tool that doesn’t necessarily make sense in a server environment. When I start the NewtorkManager, it adds 2 minutes to the boot process, but then PhotonVision foes see the connection.

mcm001, 2023-12-21:
No, Ubuntu desktop. But NetworkManager is not a GUI tool, and it absolutely shouldn't be adding 2 mins to the boot process.

mcm001, 2023-12-21:
OK yeah looks like NetworkManager is not installed by default, that sucks. I really dont want to support more than one config option though. Here's the state of things right now

- ubuntu 22.04 server seems to use netplan
- debian 12 uses ifupdown by default?
- Raspberry pi OS uses networkmanager

crschardt, 2023-12-23:
Can we update the install script to add networkmanager and nmcli? If that's the right way to do it, I can submit a PR.

mcm001, 2023-12-23:
@Bankst was scared to do it automatically, but I do want to only support NetworkManager

crschardt, 2023-12-25:
@kinahawi - I managed to get it working on a OrangePi running Ubuntu 22.04 server by the following:

1. Install network-manager: `sudo apt install network-manager`
2. Creating the file `/etc/netplan/00-default-nm-renderer.yaml` with the following contents:
```
network:
  renderer: NetworkManager
```
3. Rebooting the system.

Could you try this on your system and let me know if it fixes the problem?

kinahawi, 2023-12-27:
Sorry for the delay in responding, but the holiday got in the way.

When I try to install NetworkManager, It tells me that it is already installed. That may be because in an earlier debug attempt, I had to install “nmcli” and suspect that may have installed it implicitly.

Anyway, I created the file and rebooted and got the same result; The boot proceeds and then stops and waits for 2 minutes before moving on.

[  OK ] Reached Target Host and Network Name Lookups
[ FAILED] A Start job is running for wait for network to be configured.

Here is the relevant portion of the startup log:

Dec 27 21:06:36 photonvision systemd-resolved[636]: Using system hostname 'photonvision'.
Dec 27 21:06:36 photonvision systemd[1]: Started Network Name Resolution.
Dec 27 21:06:36 photonvision systemd[1]: Reached target Host and Network Name Lookups.
Dec 27 21:06:41 photonvision systemd[1]: systemd-rfkill.service: Deactivated successfully.
Dec 27 21:07:06 photonvision systemd[1]: systemd-fsckd.service: Deactivated successfully.
Dec 27 21:08:36 photonvision systemd-networkd-wait-online[635]: Timeout occurred while waiting for network connectivity.
Dec 27 21:08:36 photonvision systemd[1]: systemd-networkd-wait-online.service: Main process exited, code=exited, status=1/FAILURE
Dec 27 21:08:36 photonvision systemd[1]: systemd-networkd-wait-online.service: Failed with result 'exit-code'.
Dec 27 21:08:36 photonvision systemd[1]: Failed to start Wait for Network to be Configured.
Dec 27 21:08:36 photonvision systemd[1]: Starting Initial cloud-init job (metadata service crawler)...
Dec 27 21:08:37 photonvision cloud-init[642]: Cloud-init v. 23.3.3-0ubuntu0~22.04.1 running 'init' at Wed, 27 Dec 2023 21:08:37 +0000. Up 127.98 seconds.
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +++++++++++++++++++++++++++Net device info++++++++++++++++++++++++++++
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +--------+-------+-----------+-----------+-------+-------------------+
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: | Device |   Up  |  Address  |    Mask   | Scope |     Hw-Address    |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +--------+-------+-----------+-----------+-------+-------------------+
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: | enp1s0 | False |     .     |     .     |   .   | 7c:83:34:bd:45:3f |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: |   lo   |  True | 127.0.0.1 | 255.0.0.0 |  host |         .         |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: |   lo   |  True |  ::1/128  |     .     |  host |         .         |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: | wlp2s0 | False |     .     |     .     |   .   | dc:46:28:46:95:b0 |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +--------+-------+-----------+-----------+-------+-------------------+
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +++++++++++++++++++Route IPv6 info+++++++++++++++++++
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +-------+-------------+---------+-----------+-------+
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: | Route | Destination | Gateway | Interface | Flags |
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +-------+-------------+---------+-----------+-------+
Dec 27 21:08:37 photonvision cloud-init[642]: ci-info: +-------+-------------+---------+-----------+-------+
Dec 27 21:08:37 photonvision systemd[1]: Finished Initial cloud-init job (metadata service crawler).

Thanks for your help. I’m definitely out of my element on this.

Bill

crschardt, 2023-12-27:
I read back through this thread and I'm wondering if the delayed boot problem started when you followed the advice on AskUbuntu. It recommended doing:
```
cd /etc/netplan
sudo mv 00-installer-config.yaml 00-installer-config.yaml.bak
```
I didn't have to do this. Could you post the contents of your `/etc/netplan` directory?

kinahawi, 2023-12-28:
Here is everything in /etc/netplan. 

***@***.***:/etc/netplan$ ls -l
total 16
-rw-r--r-- 1 root root  38 Dec 27 21:06 00-default-nm-renderer.yaml
-rw-r--r-- 1 root root 116 Dec 19 14:27 00-enable-network-manager.yaml.bak
-rw------- 1 root root  86 Nov  2 20:36 00-installer-config-wifi.yaml
-rw-r--r-- 1 root root 118 Dec 19 21:42 00-installer-config.yaml.bak
***@***.***:/etc/netplan$ more 00-default-nm-renderer.yaml 
network:
   renderer: NetworkManager

***@***.***:/etc/netplan$ more 00-enable-network-manager.yaml.bak 
network:
  version: 2
  renderer: networkd
#  ethernets:
#     enp1s0:
#        dhcp4: true
#        optional: true
***@***.***:/etc/netplan$ more 00-installer-config.yaml.bak 
# This is the network config written by 'subiquity'
network:
  version:  2
  ethernets:
    enp1s0:
      dhcp4: true
***@***.***:/etc/netplan$ sudo more 00-installer-config-wifi.yaml 
[sudo] password for photonvision: 
# This is the network config written by 'subiquity'
network:
  version: 2
  wifis: {}

I’ve tried lots of permutations with files being live (.yaml) and deactivated (.yaml.bak) with no different results. 
I’ve tried using 00-enable-network-manager.yaml with the renderer set to NetworkManager and other times as networkd (this is from the AskUbuntu thread)
I’ve tried using 00-default-nm-renderer.yaml as you suggested.

The only combination that does not result in a delay is the original configuration of only 00-installer-config.yaml and 00-installer-config-wifi.yaml singing networkd, but then PhotonVision can’t see my network connection.

In case you need it, here is the network config:

***@***.***:~$ ifconfig
enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.68.115  netmask 255.255.255.0  broadcast 192.168.68.255
        inet6 fe80::9c15:bf91:deaa:7172  prefixlen 64  scopeid 0x20<link>
        ether 7c:83:34:bd:45:3f  txqueuelen 1000  (Ethernet)
        RX packets 2490  bytes 341340 (341.3 KB)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 453  bytes 41559 (41.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2936  bytes 174480 (174.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2936  bytes 174480 (174.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp2s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:46:28:46:95:b0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Some other background info - This system was created a couple of months back on a brand new Beeline with the 2023 release of PhotonVision. About a week ago, I copied down the JAR file for the beta and copied it n place of the 2023 version. I haven’t made any other changes.

Thanks again for the assistance.

Bill

crschardt, 2023-12-28:
Thanks for the additional information. I'm not sure what else to try. I suspect that there's something wrong with the netplan config files or that cloud-init is messing with things, but without a beelink to try it on, I'm just shooting in the dark.

kinahawi, 2023-12-29:
I finally disabled systemd.networkd-wait-online.service and that eliminated the 2 minute delay on startup. So far, no adverse affects observed.

Thanks for the support!

Bill

PR #1070 (2024-01-06): Install NetworkManager on Ubuntu distributions (fixes #1052)
This improves networking on Ubuntu distros. It provides a working nmcli that fixes the networking errors shown on OrangePi. 

This fixes #1052.
commit b8a6a5d5 (2024-01-06): Install NetworkManager on Ubuntu distributions (fixes #1052) (#1070)

Add the following args to the install script:

Syntax: sudo ./install.sh [-h|m|n|q]
  options:
  -h        Display this help message.
  -m        Install and configure NetworkManager (Ubuntu only).
  -n        Disable networking. This will also prevent installation of NetworkManager.
  -q        Silent install, automatically accepts all defaults. For non-interactive use.
closed 2024-01-06 by mcm001