PhotonVision/photonvision #1053: Updating from 2023 SQL file throws error on deserializing April Tag pipeline

crschardt, 2023-12-17:
In #981, the enum `TargetModel.k6in_16h5` was renamed to `kAprilTag6in_16h5`. When trying to update the settings stored in photonvision.sqlite, this causes the deserialization of the pipeline to fail and the pipeline reverts to reflective. 

** Exception **
```
[2023-12-17 11:17:33] [General - LegacyConfigProvider] [ERROR] Exception while deserializing cameras/Arducam_OV9281_USB_Camera/pipelines/New Pipeline.json: Cannot deserialize value of type `org.photonvision.vision.target.TargetModel` from String "k6in_16h5": not one of the values accepted for Enum class: [k2016HighGoal, kAprilTag6in_16h5, kAprilTag6p5in_36h11, k2019DualTarget, k2020HighGoalOuter, k2022CircularCargoBall, kCircularPowerCell7in]
 at [Source: (File); line: 66, column: 19] (through reference chain: org.photonvision.vision.pipeline.AprilTagPipelineSettings["targetModel"])
[2023-12-17 11:17:33] [General - LegacyConfigProvider] [ERROR] com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `org.photonvision.vision.target.TargetModel` from String "k6in_16h5": not one of the values accepted for Enum class: [k2016HighGoal, kAprilTag6in_16h5, kAprilTag6p5in_36h11, k2019DualTarget, k2020HighGoalOuter, k2022CircularCargoBall, kCircularPowerCell7in]
 at [Source: (File); line: 66, column: 19] (through reference chain: org.photonvision.vision.pipeline.AprilTagPipelineSettings["targetModel"])
        at com.fasterxml.jackson.databind.exc.InvalidFormatException.from(InvalidFormatException.java:67)
        at com.fasterxml.jackson.databind.DeserializationContext.weirdStringException(DeserializationContext.java:2002)
        at com.fasterxml.jackson.databind.DeserializationContext.handleWeirdStringValue(DeserializationContext.java:1230)
        at com.fasterxml.jackson.databind.deser.std.EnumDeserializer._deserializeAltString(EnumDeserializer.java:415)
        at com.fasterxml.jackson.databind.deser.std.EnumDeserializer._fromString(EnumDeserializer.java:279)
        at com.fasterxml.jackson.databind.deser.std.EnumDeserializer.deserialize(EnumDeserializer.java:248)
        at com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:138)
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:314)
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:177)
        at com.fasterxml.jackson.databind.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:120)
        at com.fasterxml.jackson.databind.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromObject(AsArrayTypeDeserializer.java:61)
        at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.deserializeWithType(BeanDeserializerBase.java:1296)
        at com.fasterxml.jackson.databind.deser.impl.TypeWrappedDeserializer.deserialize(TypeWrappedDeserializer.java:74)
        at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:323)
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4825)
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3637)
        at org.photonvision.common.util.file.JacksonUtils.deserialize(JacksonUtils.java:88)
        at org.photonvision.common.configuration.LegacyConfigProvider.lambda$loadCameraConfigs$2(LegacyConfigProvider.java:339)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
        at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
        at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
        at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
        at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:913)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
        at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:578)
        at org.photonvision.common.configuration.LegacyConfigProvider.loadCameraConfigs(LegacyConfigProvider.java:351)
        at org.photonvision.common.configuration.LegacyConfigProvider.load(LegacyConfigProvider.java:213)
        at org.photonvision.common.configuration.ConfigManager.translateLegacyIfPresent(ConfigManager.java:96)
        at org.photonvision.common.configuration.ConfigManager.load(ConfigManager.java:183)
        at org.photonvision.Main.main(Main.java:348)
```

mcm001, 2023-12-17:
Does Jackson have support for deserialzing enums that have changed names? If not, we could just make 2024 a hard break and say everything from 2023 won't work anymore?

crschardt, 2023-12-17:
It turns out that yes, there is a simple way to fix this. The PR is #1055.

mcm001, 2023-12-18:
Closed by #1055

closed 2023-12-18 by mcm001