PhotonVision/photonvision #1057: Install Script can't install Java when using bookworm

Technologyman00, 2023-12-18:
**Describe the bug**
The install script can't install Java when using bookworm 

This package doesn't exist in apt for bookworm openjdk-11-jre-headless

**To Reproduce**
Use install scriptnon bookworm
Won't boot photonvision because no java

**Platform:**
 - Hardware Platform: Pi 5

**Additional context**
Can be overcome by installing default-jdk

mcm001, 2023-12-19:
Swap this to 17 and report back? If so make a PR to change the install script over to 17?

BytingBulldogs3539, 2023-12-19:
> Swap this to 17 and report back? If so make a PR to change the install script over to 17?

Swapping to 17 fixed this for me, guess I forgot to update that.

srimanachanta, 2024-01-01:
@Technologyman00 Does the new install script work for you? Can this be closed?

Technologyman00, 2024-01-01:
Yep I just tried it can confirm it works

closed 2024-01-01 by Technologyman00