PhotonVision/photonvision #1060: Photonvision bootloops when more than 1 CSI camera is connected on Raspberry Pi 5 / Compute Module 4

SeanErn, 2023-12-20:
**Describe the bug**
On both the beta branches and stable, Photonvision crashes repeatedly when I have 2 CSI cameras connected. I am testing this on a Compute Module 4. Sometimes, it launches and only one camera is detected, then crashes back into the loop again after ~20 seconds. I think the error is related to this (found in syslog):
```
Jan 31 20:32:10 photonvision java[6190]: [0:05:49.570638142] [6296] #033[1;31mERROR #033[1;37mV4L2 #033[1;34mv4l2_device.cpp:359 #033[0;32m'ov9281 0-0060': #033[0mUnable to set control 0x00980911: Input/output error
Jan 31 20:32:10 photonvision java[6190]: [0:05:49.577598698] [6296] #033[1;31mERROR #033[1;37mV4L2 #033[1;34mv4l2_videodevice.cpp:1906 #033[0;32m/dev/video0[38:cap]: #033[0mFailed to start streaming: Input/output error
Jan 31 20:32:10 photonvision java[6190]: terminate called after throwing an instance of 'std::runtime_error'
Jan 31 20:32:10 photonvision java[6190]:   what():  failed to start camera
Jan 31 20:32:10 photonvision kernel: [  349.578250] ov9281 0-0060: start stream failed while write regs
Jan 31 20:32:10 photonvision kernel: [  349.578285] unicam fe800000.csi: stream on failed in subdev
Jan 31 20:32:10 photonvision systemd[1]: photonvision.service: Main process exited, code=killed, status=6/ABRT
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install two identical OV9281 cameras (have not tested them with other cameras, but these are directly from the recommended list)
2. Connect the board to power and Ethernet
3. Wait for Photonvision to boot and run/crash for 1 minute
4. Try to open web dashboard and fail
5. See errors in rpi os syslog

**Screenshots / Videos**
I don't have photon logs as it crashes before the logger starts. But here is my syslog and config
Syslog:
```
Jan 31 20:32:03 photonvision systemd[1]: Started PhotonVision server.
Jan 31 20:32:06 photonvision java[6190]: #033[32m[2023-01-31 20:32:06] [General - Main] [INFO] Native libraries loaded.#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[32m[2023-01-31 20:32:06] [Camera - LibCameraJNI] [INFO] Successfully loaded libpicam shared object#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[32m[2023-01-31 20:32:06] [General - Main] [INFO] Logging initialized in debug mode.#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[37m[2023-01-31 20:32:06] [General - ShellExec] [DEBUG] Executing "cat /proc/device-tree/model"#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[37m[2023-01-31 20:32:06] [General - ShellExec] [DEBUG] Got exit code 0#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[32m[2023-01-31 20:32:06] [General - Main] [INFO] Starting PhotonVision version v2024.1.1-beta-3.2 on Linux Raspbian 64-bit (Pi UNKNOWN)#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[37m[2023-01-31 20:32:06] [General - Main] [DEBUG] Loading ConfigManager...#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[37m[2023-01-31 20:32:06] [Config - SqlConfigProvider] [DEBUG] Loading config...#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[31m[2023-01-31 20:32:06] [Config - SqlConfigProvider] [ERROR] Could not deserialize apriltag layout! Loading defaults#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[37m[2023-01-31 20:32:06] [General - Main] [DEBUG] Loading HardwareManager...#033[0m
Jan 31 20:32:06 photonvision java[6190]: #033[32m[2023-01-31 20:32:06] [General - NetworkTablesManager] [INFO] Starting NT Client, server team is 5152#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[31m[2023-01-31 20:32:07] [General - NetworkTablesManager] [ERROR] [NetworkTablesManager] Could not connect to the robot! Will retry in the background...#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - Main] [DEBUG] Loading NetworkManager...#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[32m[2023-01-31 20:32:07] [General - NetworkManager] [INFO] Setting STATIC with team 5152#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Executing "cat /etc/hostname | tr -d " \t\n\r""#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Got exit code 0#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Executing "echo photonvision > /etc/hostname"#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Got exit code 0#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Executing "hostnamectl set-hostname photonvision"#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Got exit code 0#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Executing "sed -i "s/127.0.1.1.*photonvision/127.0.1.1\tphotonvision/g" /etc/hosts"#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Got exit code 0#033[0m
Jan 31 20:32:07 photonvision java[6190]: #033[37m[2023-01-31 20:32:07] [General - ShellExec] [DEBUG] Executing "sudo service avahi-daemon restart"#033[0m
Jan 31 20:32:07 photonvision avahi-daemon[6128]: Got SIGTERM, quitting.
Jan 31 20:32:07 photonvision avahi-daemon[6128]: Leaving mDNS multicast group on interface eth0.IPv4 with address 10.51.52.11.
Jan 31 20:32:07 photonvision systemd[1]: Stopping Avahi mDNS/DNS-SD Stack...
Jan 31 20:32:07 photonvision avahi-daemon[6128]: Leaving mDNS multicast group on interface lo.IPv6 with address ::1.
Jan 31 20:32:07 photonvision avahi-daemon[6128]: Leaving mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Jan 31 20:32:07 photonvision avahi-daemon[6128]: avahi-daemon 0.8 exiting.
Jan 31 20:32:07 photonvision systemd[1]: avahi-daemon.service: Succeeded.
Jan 31 20:32:07 photonvision systemd[1]: Stopped Avahi mDNS/DNS-SD Stack.
Jan 31 20:32:07 photonvision systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Process 6128 died: No such process; trying to remove PID file. (/run/avahi-daemon//pid)
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Found user 'avahi' (UID 108) and group 'avahi' (GID 112).
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Successfully dropped root privileges.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: avahi-daemon 0.8 starting up.
Jan 31 20:32:07 photonvision systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Successfully called chroot().
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Successfully dropped remaining capabilities.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: No service file found in /etc/avahi/services.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Joining mDNS multicast group on interface eth0.IPv4 with address 10.51.52.11.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: New relevant interface eth0.IPv4 for mDNS.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: New relevant interface lo.IPv6 for mDNS.
Jan 31 20:32:07 photonvision avahi-daemon[6253]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
[... 163 lines cut ...]
##############################################################
### PHOTONVISION CAM CONFIG
### Comment/Uncomment to change which camera is supported
### Picam V1, V2 or HQ: uncomment (remove leading # ) from camera_auto_detect=1,
### and comment out all following lines
### IMX290/327/OV9281/Any other cameras that require additional overlays:
### Comment out (add a # ) to camera_auto_detect=1, and uncomment the line for
### the sensor you're trying to user

# cameraAutoDetect=1

# dtoverlay=imx290,clock-frequency=74250000
# dtoverlay=imx290,clock-frequency=37125000
# dtoverlay=imx378
# dtoverlay=ov9281

##############################################################
```

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Raspberry Pi CM4 with [CM4-IO-BASE-A](https://www.waveshare.com/wiki/CM4-IO-BASE-A)
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Tested with P2P, 2nd router port, and switch
 - PhotonVision Version: Tested with 2024-beta-6 and 2023.4.2
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Firefox 119.1
 - Camera(s) Used: 2x OV9281 CSI

mcm001, 2023-12-20:
Also @SeanErn are you on Discord? We've got another contributor interested in supporting dual cam. Suspect this will be a good chunk of C++/native code, would be good to connect in a more real time context

SeanErn, 2023-12-20:
> Also @SeanErn are you on Discord? 

My discord is `funky.chicken`  and I'd be happy to connect with them any time. I have school holiday break coming up from the end of this week to Kickoff so I am pretty open.

Also, you are correct that it works fine with one camera.

SeanErn, 2023-12-24:
This is fixed with the new PR #1068

closed 2023-12-24 by SeanErn