PhotonVision/photonvision #1063: Remove LegacyConfigProvider

srimanachanta, 2023-12-20:
The LegacyConfig provider should be removed. The settings provider should check for legacy files on startup, convert them into SQL if they exist, and then delete the legacy files. This way only the SQL config provider exists

mcm001, 2023-12-20:
That's... what happens right now? I'm confused how what you propose is any different

srimanachanta, 2023-12-21:
> That's... what happens right now? I'm confused how what you propose is any different

My bad, I mean more like instead of having a legacy config provider, add a check in the creation of the ConfigProvider to convert it and then we don't have to have the ConfigProvider and all that stuff just to have it be replaced

mcm001, 2023-12-21:
Still unclear on what you mean about replacing?

closed 2024-01-01 by srimanachanta