PhotonVision/photonvision #1072: getLatestsResult gives an ArrayIndexOutOfBoundsException, and sometimes the photonvision frontend is loading without showing anything.

levyishai, 2023-12-26:
**Describe the bug**
When using the getLatestResult function in java, with the latest vendordep, the following exception appears:
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception in Notifier thread: java.lang.ArrayIndexOutOfBoundsException: Index 914 out of bounds for length 914 ﻿﻿ org.photonvision.common.dataflow.structures.Packet.decodeByte(Packet.java:145) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at org.photonvision.common.dataflow.structures.Packet.decodeByte(Packet.java:145): Unhandled exception in Notifier thread: java.lang.ArrayIndexOutOfBoundsException: Index 914 out of bounds for length 914 ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.common.dataflow.structures.Packet.decodeByte(Packet.java:145) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.targeting.PhotonTrackedTarget.decodeList(PhotonTrackedTarget.java:208) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.targeting.PhotonTrackedTarget.createFromPacket(PhotonTrackedTarget.java:243) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.targeting.PhotonPipelineResult.createFromPacket(PhotonPipelineResult.java:170) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.PhotonCamera.getLatestResult(PhotonCamera.java:198) ﻿

Additionally, sometimes, the photonvision client just loads and doesn't show anything with the following problem in the system log: 
Dec 08 11:37:48 photonvision java[3243]: [2023-12-08 11:37:48] [WebServer - Server] [DEBUG] Handled HTTP request of type GET from endpoint /assets/loading-0f0dc28b.svg for host 10.59.90.221 in 1.005646 ms
Dec 08 11:37:48 photonvision java[3243]: [2023-12-08 11:37:48] [WebServer - Server] [DEBUG] Handled HTTP request of type GET from endpoint /assets/logoLarge-7c0b93cf.svg for host 10.59.90.221 in 2.909024 ms
Dec 08 11:37:48 photonvision java[3243]: [2023-12-08 11:37:48] [WebServer - Server] [DEBUG] Handled HTTP request of type GET from endpoint /assets/prompt-latin-400-normal-b7be1a25.woff2 for host 10.59.90.221 in 1.534719 ms
Dec 08 11:37:48 photonvision java[3243]: [2023-12-08 11:37:48] [WebServer - Server] [DEBUG] Handled HTTP request of type GET from endpoint /assets/materialdesignicons-webfont-c1c004a9.woff2 for host 10.59.90.221 in 1.544052 ms
Dec 08 11:37:49 photonvision java[3243]: [2023-12-08 11:37:49] [WebServer - Server] [DEBUG] Handled HTTP request of type GET from endpoint /favicon.ico for host 10.59.90.221 in 2.929731 ms

**To Reproduce**
Steps to reproduce the behavior:
Use photonCamera.getLatestResult();
idk why the second problem occurs. 

**Screenshots / Videos**
This is the image of loading without anything happening: 
[image]
This is the system log: 
[photonvisionlog.txt](https://github.com/PhotonVision/photonvision/files/13773275/photonvisionlog.txt)

**Platform:**
 - Hardware Platform:  OrangePi 5 running Ubuntu.
 - Network Configuration: An OV9281 camera connected via USB to an orangepi, connected straight to a router, with a roborio 2.0 connected to the second port of the router.
 - PhotonVision Version: The orange pi is running the `photonvision-dev-v2024.1.1-beta-3.2-27-g0b98f027-linuxarm64.jar`. And using the `v2024.1.1-beta-3.2` vendordep. 
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome 120.0.6099.129
 - Camera(s) Used: OV9281

mcm001, 2023-12-26:
You need to be using the same vendordep version and photonvision version, so I'm not shocked about the array out of bounds. You should have seen a warning earlier in your logs telling you about the version mismatch as well.

levyishai, 2023-12-26:
> You need to be using the same vendordep version and photonvision version, so I'm not shocked about the array out of bounds. You should have seen a warning earlier in your logs telling you about the version mismatch as well.

Wait, is there a newer version for the vendordep? 
Vs code didn't show anything

mcm001, 2023-12-26:
Yeah, vscode won't show it but we have a beta vendordep. Replace the version string with the version of photon on your Coproc

levyishai, 2023-12-26:
> Yeah, vscode won't show it but we have a beta vendordep. Replace the version string with the version of photon on your Coproc

I changed the version and a similar issue occurs. 
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception in Notifier thread: java.lang.ArrayIndexOutOfBoundsException: Index 562 out of bounds for length 562 ﻿﻿ org.photonvision.common.dataflow.structures.Packet.decodeDouble(Packet.java:172) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at org.photonvision.common.dataflow.structures.Packet.decodeDouble(Packet.java:172): Unhandled exception in Notifier thread: java.lang.ArrayIndexOutOfBoundsException: Index 562 out of bounds for length 562 ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.common.dataflow.structures.Packet.decodeDouble(Packet.java:172) ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The Runnable for this Notifier (or methods called by it) should have handled the exception above.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information. ﻿﻿ edu.wpi.first.wpilibj.Notifier.lambda$new$1(Notifier.java:138) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.utils.PacketUtils.decodeTransform(PacketUtils.java:33) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.targeting.PhotonTrackedTarget.createFromPacket(PhotonTrackedTarget.java:232) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.targeting.PhotonPipelineResult.createFromPacket(PhotonPipelineResult.java:170) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.photonvision.PhotonCamera.getLatestResult(PhotonCamera.java:198)

mcm001, 2023-12-27:
[image]

Not able to reproduce with my vendordep version and photon version set to ` dev-v2024.1.1-beta-3.2-27-g0b98f027 `

closed 2023-12-29 by mcm001