PhotonVision/photonvision #1085: getLatestsResult gives an ArrayIndexOutOfBoundsException when using PhotonLib v2024.1.1-beta-3.2 with OrangePi running PhotonVision v2023.3.0

dsawdai, 2023-12-29:
**Describe the bug**
Java robot code immediately crashes when using the 2024 beta PhotonLib, if the RoboRio is connected to an OrangePi running 2023 PhotonVision.  More specifically, getLatestsResult gives an ArrayIndexOutOfBoundsException when using PhotonLib v2024.1.1-beta-3.2 with OrangePi running PhotonVision v2023.3.0.

**Screenshots / Videos**
See console trace:
Photon version v2024.1.1-beta-3.2 does not match coprocessor version v2023.3.0!
Warning at org.photonvision.PhotonCamera.verifyVersion(PhotonCamera.java:434): Photon version v2024.1.1-beta-3.2 does not match coprocessor version v2023.3.0!
at org.photonvision.PhotonCamera.verifyVersion(PhotonCamera.java:434)
at org.photonvision.PhotonCamera.getLatestResult(PhotonCamera.java:186)
at org.photonvision.PhotonPoseEstimator.update(PhotonPoseEstimator.java:305)
at frc.robot.subsystems.PhotonCameraWrapper.getEstimatedGlobalPose(PhotonCameraWrapper.java:116)
at frc.robot.subsystems.DriveTrain.updateOdometry(DriveTrain.java:601)
at frc.robot.subsystems.DriveTrain.periodic(DriveTrain.java:533)
at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:260)
at frc.robot.Robot.robotPeriodic(Robot.java:53)
at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:372)
at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:139)
at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:361)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:453)
at frc.robot.Main.main(Main.java:23)

Unhandled exception: java.lang.ArrayIndexOutOfBoundsException: Index 10 out of bounds for length 10
Error at org.photonvision.common.dataflow.structures.Packet.decodeShort(Packet.java:214): Unhandled exception: java.lang.ArrayIndexOutOfBoundsException: Index 10 out of bounds for length 10
at org.photonvision.common.dataflow.structures.Packet.decodeShort(Packet.java:214)
at org.photonvision.targeting.MultiTargetPNPResults.createFromPacket(MultiTargetPNPResults.java:44)
at org.photonvision.targeting.PhotonPipelineResult.createFromPacket(PhotonPipelineResult.java:162)
at org.photonvision.PhotonCamera.getLatestResult(PhotonCamera.java:198)
at org.photonvision.PhotonPoseEstimator.update(PhotonPoseEstimator.java:305)
at frc.robot.subsystems.PhotonCameraWrapper.getEstimatedGlobalPose(PhotonCameraWrapper.java:116)
at frc.robot.subsystems.DriveTrain.updateOdometry(DriveTrain.java:601)
at frc.robot.subsystems.DriveTrain.periodic(DriveTrain.java:533)
at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:260)
at frc.robot.Robot.robotPeriodic(Robot.java:53)
at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:372)
at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:139)
at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:361)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:453)
at frc.robot.Main.main(Main.java:23)

The robot program quit unexpectedly. This is usually due to a code error.
The above stacktrace can help determine where the error occurred.
See https://wpilib.org/stacktrace for more information.
Warning at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:375): The robot program quit unexpectedly. This is usually due to a code error.
The above stacktrace can help determine where the error occurred.
See https://wpilib.org/stacktrace for more information.
The startCompetition() method (or methods called by it) should have handled the exception above.
Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:382): The startCompetition() method (or methods called by it) should have handled the exception above.
[phoenix-diagnostics] Server shutdown cleanly. (dur:22)
[phoenix] Library shutdown cleanly
#
# A fatal error has been detected by the Java Runtime Environment:
#
# SIGSEGV (0xb) at pc=0xb6dcdc44, pid=2656, tid=2660
#
# JRE version: OpenJDK Runtime Environment (17.0.9.7) (build 17.0.9.7-frc+0-2024-17.0.9u7-1)
# Java VM: OpenJDK Client VM (17.0.9.7-frc+0-2024-17.0.9u7-1, mixed mode, emulated-client, serial gc, linux-arm)
# Problematic frame:
# C [libc.so.6+0x6ac44]
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /tmp/hs_err_pid2656.log
java: tpp.c:82: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.

**Platform:**
 - Hardware Platform:  Orange Pi, RoboRio, Windows x64
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch):  Radio port 1 to RoboRio, Radio port 2 to network switch.  Network switch to OrangePi and to driver station laptop (Windows x64).
 - PhotonVision Version:  PhotonLib v2024.1.1-beta-3.2, PhotonVision v2023.3.0

mcm001, 2023-12-29:
Duplicate of #1072

closed 2023-12-29 by mcm001

mcm001, 2023-12-29:
Closing for now, but reopen if you continue seeing it