PhotonVision/photonvision #1086: Update build system for examples

srimanachanta, 2023-12-29:
The build system for examples should be refactored to have java and cpp common build fixes for each subproject so that each example's dependencies are better tied to the current versions and state of the monorepo. Currently, each subproject has to have its version explicitly specified which is redundant and may lead to further confusion to people looking to use the example beyond just running it.

I belive an attempt at this was made earlier (years ago) but was only partly successful so more discussion may be worth it.

mcm001, 2023-12-29:
It's arguably good to leverage the vendordep system. We get a full end to end gradleRio test for free basically. But it was nice when they were integrated.

mcm001, 2024-11-02:
Closing as an opinionated decision to keep example projects able to be copy-pasted or opened in WPILib VS Code. This isnt 100% possible yet but I'd argue the current structure is Correct.

closed 2024-11-02 by mcm001