PhotonVision/photonvision #1092: The AprilTag settings tab doesn't update when switching between cameras in a multi-camera system.

crschardt, 2023-12-30:
**Describe the bug**
When using a coprocessor with two cameras, both set to AprilTags pipelines, switching from one camera to the other doesn't update the settings shown on the AprilTags tab in the UI. The Input and Output tabs update correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach two cameras to a coprocessor (I'm using two Aruducam OV9281s on an Orange Pi 5)
2. Configure an AprilTag pipeline for the first camera.
3. Switch to the second camera and select the AprilTag pipeline
4. The settings under the AprilTag tab will be identical to what was set for the first camera.
5. Change the settings for the second camera to be different from the first ones.
6. Switch back to the first camera and the displayed settings will remain the settings for the second camera.

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Connected to my network switch
 - PhotonVision Version: dev-v2024.1.1-beta-3.2-33-g033c3616
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Edge
 - Camera(s) Used: 2x OV9281, with different camera names (renamed with Arducam tool)

**Additional context**
As far as I can tell, the pipeline settings for the two cameras are different. The bug is just in updating the values displayed in the UI.

randomstring, 2023-12-30:
I could re-create with a single camera setup. 

Duplicate a pipeline, change a setting (for instance change decimate 2  to 3) in the new pipeline. When you switch between pipelines in the UI, the changed setting doesn't update in the UI.

mcm001, 2023-12-30:
Yeah I'm seeing the same thing. The backend settings seem to be correct, but the frontend display doesn't update to match

mcm001, 2023-12-30:
Caused by/duplicate of https://github.com/PhotonVision/photonvision/issues/1030 -- let's move there

closed 2023-12-30 by mcm001