PhotonVision/photonvision #1093: Documentation missing for several PhotonLib parameters

calcmogul, 2023-12-30:
CI shows several warnings for missing docs.
https://github.com/PhotonVision/photonvision/actions/runs/7353788025/job/20033628834#step:5:76

Consider setting the flag that promotes doc warnings to errors for JavaDoc and Doxygen to catch these in the future.

srimanachanta, 2024-01-01:
Ditto on this, I have a branch with a bunch of docs done but the warning should be promoted for all user-facing javadocs

samfreund, 2025-04-12:
Bumping this, I think we should prioritize it for this offseason.

samfreund, 2025-04-16:
So upon further investigation, this defenitely needs to be a priority, but there's a crap-ton of files that don't have docs. It's gonna take a while to work thru all of them.