PhotonVision/photonvision #1111: Orange Pi 5 Build and Documentation

dgreathouse, 2024-01-04:
**Describe the bug**
Documentation for Orange Pi 5 software installation states to download the "orangepi5.xz" file from the releases page.
There is no "orangepi5.xz" file on the release page under Assets. 

**To Reproduce**
Goto [https://docs.photonvision.org/en/latest/docs/installation/sw_install/orange-pi.html](url) for the documentation 
See [https://github.com/PhotonVision/photonvision/releases](url) for the release Assets.

**Screenshots / Videos**

**Platform:**
Orange Pi 5 for PhotonVision

**Additional context**

mcm001, 2024-01-04:
Looks like the docs PR was merged before https://github.com/PhotonVision/photonvision/pull/1081 . 

Closed by https://github.com/PhotonVision/photonvision/pull/1081

mcm001, 2024-01-04:
https://github.com/PhotonVision/photonvision/releases/tag/Dev

This release now has an orange pi image -- any chance you can confirm it works?

dgreathouse, 2024-01-05:
The orange PI 5 boots as expected and can be accessed through the address. The Dashboard is visible and functioning. Tried a couple of unsupported USB cameras and my Nexigo_N60_FHD camera worked as expected. Driver view and pipeline work with the camera. Trying to work the process to get the image to boot from the NVMe card and will post links or procedure for teams that may want to remove the reliance on a SD Card.

mcm001, 2024-01-05:
> The orange PI 5 boots as expected and can be accessed through the address. The Dashboard is visible and functioning. Tried a couple of unsupported USB cameras and my Nexigo_N60_FHD camera worked as expected. Driver view and pipeline work with the camera. Trying to work the process to get the image to boot from the NVMe card and will post links or procedure for teams that may want to remove the reliance on a SD Card. SSH works as expected. 

Awesome, good to hear! If you figure out the NVMe process feel free to write that up and contribute it to our docs :) . I'm also not a fan of SD cards

mcm001, 2024-11-02:
Closing as stale

closed 2024-11-02 by mcm001