PhotonVision/photonvision #1114: Mulitag samples aren't properly filling in Client

srimanachanta, 2024-01-04:
The sample count should be increasing with each new result but it isn't. See here, sample count is stuck at 8.

https://github.com/PhotonVision/photonvision/assets/68172138/7d0336ab-09cd-4f3d-8719-ccf1b5db0790

randomstring, 2024-01-05:
I've seen this a couple times.  Samples will stop updating.  Reports NaN/100 samples.

I'll add that hitting the "RESET SAMPLES" button will sometimes lock up the UI, Firefox reports that the web page is using too much cpu. Only fix is to close the tab and reconnect.

srimanachanta, 2024-01-08:
I belive the fact that we are using a while loop to remove samples which the store is adding them is bad practice and relys on getting more samples then are removed, it should be redone. Not sure why samples are filling in but might also be related to that.