PhotonVision/photonvision #1129: When changing hostname, the previous hostname is set instead

mcm001, 2024-01-08:
After fully rebooting the Pi with hostname set to photonvision-2: boot logs show was set to photonvision-2 at boot
After changing in web ui to photonvision-3: logs show `Jan 08 01:23:41 photonvision-2 java[728]: [2024-01-08 01:23:41] [General - NetworkManager] [INFO] Set hostname to photonvision-2`
After changing in web ui to photonvision-4: logs show `Jan 08 01:23:41 photonvision-2 java[728]: [2024-01-08 01:23:41] [General - NetworkManager] [INFO] Set hostname to photonvision-3`
After changing in web ui to photonvision-2: logs show `Jan 08 01:23:41 photonvision-2 java[728]: [2024-01-08 01:23:41] [General - NetworkManager] [INFO] Set hostname to photonvision-4`

mcm001, 2024-01-08:
Same bug is present with static IP configuration -- changing from 192.168.1.201 to 192.168.1.202 and then sending save sends 201 instead of 202 to the backend

mcm001, 2024-01-08:
Same thing happens with team number/NT address entry box...

BytingBulldogs3539, 2024-01-08:
Same thing happens with run networktables server toggle

PR #1131 (2024-01-08): Fix bug with saving general settings not using tempSettingsStruct and using store values instead
residual from #1075 
closes #1129
commit 4d458198 (2024-01-08): Fix bug with saving general settings not using tempSettingsStruct and using store values instead (#1131)

residual from #1075
closes #1129
closed 2024-01-08 by mcm001