PhotonVision/photonvision #1132: Pi Cam V1

BytingBulldogs3539, 2024-01-08:
**Describe the bug**
Pi Cam v1 will show the correct color for a short time then end up turning green, blue, or yellow.

**To Reproduce**
View Pi cam v1 on rpi5 at 320p or 480p.

**Screenshots / Videos**
[image]

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): RPI 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Home Network
 - PhotonVision Version:    dev-v2024.1.1-beta-4.2-7-g4696943f
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Pi cam v1

**Additional context**
Latest photonvision RPI image.

https://github.com/raspberrypi/libcamera/issues/87 "Changes should be merged soon. I'll resolve this now." still not merged to public builds..

mcm001, 2024-01-08:
Looks like it was fixed in https://github.com/raspberrypi/libcamera/commit/2fae9603e6cc483d9d0d74868721b272776513cf which is on their next branch still

BytingBulldogs3539, 2024-03-03:
Looks like libcamera pulled this change into their main branch this should be revisited and see if it is fixed.

closed 2024-06-17 by Juniormunk