PhotonVision/photonvision #1142: Orange Pi 5 Image installs v2023.4.2 instead of latest

judsonjames, 2024-01-08:
**Describe the bug**
Orange Pi 5 image correctly installs PhotonVision and Ubuntu 22.04.3 LTS, however, instead of installing the latest version of PhotonVision (as of now, version 2024.1.2), it installs version 2023.4.2.

**To Reproduce**
Steps to reproduce the behavior:
1. Download PhotonVision image (v2024.1.2) for OrangePi5 
2. Image SD Card (FWIW, I'm using Etcher)
3. Insert SD Card to OrangePi 5
4. Turn on OrangePi 5
5. Open settings tab of PhotonVision's web dashboard and see incorrect version (and everything else being the previous version)

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): OrangePi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): network switch on isolate network, not connected to RIO
 - PhotonVision Version: v2024.1.2 (supposed to be)
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Arducam OV9281

**Additional context**
Every other step of installation and connecting to Orange Pi 5 is accurate. If I SCP the ARM image manually and overwrite the service JAR, it runs exactly as intended.

mcm001, 2024-01-08:
[image]

Baboom. The bug. We should swap to arm runner for this

srimanachanta, 2024-01-14:
Can this be closed?

judsonjames, 2024-01-14:
My latest PR resolves this. This can be closed.

closed 2024-01-14 by judsonjames