PhotonVision/photonvision #1145: photonlibpy fails type check

auscompgeek, 2024-01-10:
**Describe the bug**
```
photonlibpy/photonCamera.py:89: error: Argument 1 to "Packet" has incompatible type "bytes"; expected "list[int]"  [arg-type]
```

**To Reproduce**
```sh
pip install mypy
mypy photonlib/
```

auscompgeek, 2024-01-28:
More errors now!

```
photonlibpy/estimatedRobotPose.py:22: error: Bracketed expression "[...]" is not valid as a type  [valid-type]
photonlibpy/estimatedRobotPose.py:22: note: Did you mean "List[...]"?
photonlibpy/photonPoseEstimator.py:151: error: Incompatible return value type (got "Pose3d | None", expected "Pose3d")  [return-value]
photonlibpy/photonPoseEstimator.py:167: error: Incompatible return value type (got "Pose3d | None", expected "Pose3d")  [return-value]
photonlibpy/photonPoseEstimator.py:207: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:211: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:220: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:223: error: Incompatible types in assignment (expression has type "float", variable has type "int")  [assignment]
photonlibpy/photonPoseEstimator.py:227: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:242: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:290: error: Incompatible types in assignment (expression has type "float", variable has type "int")  [assignment]
photonlibpy/photonPoseEstimator.py:296: error: Return value expected  [return-value]
photonlibpy/photonPoseEstimator.py:304: error: Return value expected  [return-value]
```

> Do you have bandwidth to add mypi to our CI workflow and fix all the errors?

No.

LucienMorey, 2024-11-20:
Fyy, like #1210, the last of this was solved with #1570